olikraus/u8g2 #122: Example

iotexpert, 2016-12-24:
I think that you should consider making an example directory to enable us to contribute examples.

Perhaps:

example/{c, cpp}/platform/examplename  or something.

Alan

olikraus, 2016-12-24:
this is already there:
https://github.com/olikraus/u8g2/tree/master/sys
Idea is to have this:
https://github.com/olikraus/u8g2/tree/master/sys/platform
Then:
https://github.com/olikraus/u8g2/tree/master/sys/platform/examplename
Or
https://github.com/olikraus/u8g2/tree/master/sys/platform/u8x8/examplename
and
https://github.com/olikraus/u8g2/tree/master/sys/platform/u8g2/examplename

Not sure whether it makes sense to distinguish between c and c++

olikraus, 2016-12-24:
I always wanted to start something with the psoc 4 family... But it looks very much windows oriented and i did not see how to operate from linux. But i actually never went too much into details.

iotexpert, 2016-12-24:
I ran the software division at Cypress for 3 years... and still work for Cypress... You can run PSoC Creator in a virtual box... but there is way to much invested in C# to switch.  We have started really talking about the future of Creator with the new PSoC coming... but it will be a while before you see something new.

closed 2016-12-24 by iotexpert