olikraus/u8g2 #123: Support for Adafruit St7565

fcd, 2016-12-24:
It appears the Adafruit version of the ST7565 is not supported.  This was the U8GLIB_LM6059 constructor in U8glib.  The U8G2_ST7565_EA_DOGM128_1_4W_SW_SPI starts rendering the display about halfway down the screen.

olikraus, 2016-12-24:
Can you send me a picture, please. Unfortunately i do not own this display and the u8glib code was partly contributed by others. But with your help, it should be possible to make it run again. Thanks.

fcd, 2016-12-25:
[image]

The code that generates this is: 
`    glcd.setFont(u8g2_font_ncenB14_tr);
    glcd.drawStr(0,20,"Hello World!");
    glcd.drawStr(0,40,"Hello World!");`

fcd, 2016-12-25:
I fixed the offset by changing the display start line in  u8x8_d_st7565_dogm128_init_seq to 0x060.  I also changed the LCD Bias to 0x0a3 (which fixed having to increase the contrast).  Now I seem to be shifted a little in x and then there are some "random" pixels on the right side of the display.  I'll add the display and create a pull request...I assume u8x8_d_st7565 is the correct file?

olikraus, 2016-12-25:
Great findings so far. However may plan is to create a new device for this display.
I will do the this as soon as possible.

fcd, 2016-12-26:
Ok I adjusted a few more things...here is the complete init:
`  U8X8_START_TRANSFER(),             	/* enable chip, delay is part of the transfer start */

  U8X8_C(0x0e2),            			/* soft reset */
  U8X8_C(0x0ae),		                /* display off */
  U8X8_C(0x060),		                /* set display start line to 0 */

  U8X8_C(0x0a0),		                /* ADC set to reverse */
  U8X8_C(0x0c8),		                /* common output mode */
  // Flipmode
  // U8X8_C(0x0a0),		                /* ADC set to reverse */
  // U8X8_C(0x0c8),		                /* common output mode */

  U8X8_C(0x0a6),		                /* display normal, bit val 0: LCD pixel off. */
  U8X8_C(0x0a3),		                /* LCD bias 1/9 */
  U8X8_C(0x02f),		                /* all power  control circuits on */
  U8X8_CA(0x0f8, 0x000),		/* set booster ratio to 4x */
  U8X8_C(0x027),		                /* regulator, booster and follower */
  U8X8_CA(0x081, 0x018),		/* set contrast, contrast value, EA default: 0x016 */
  U8X8_C(0x0a4),		                /* normal display  */

  U8X8_END_TRANSFER(),             	/* disable chip */
  U8X8_END()             			/* end of sequence */`

0 in x is off the display, 1 is on.  0 in Y i son the display

olikraus, 2016-12-26:
I have just created the LM6059 constructor, based on your input.
Can you check, whether this will work?
You can download the Arduino lib here:
https://github.com/olikraus/U8g2_Arduino/archive/master.zip
You can manually install this lib via the Arduino IDE (zip library import).
Constructor name is for example
U8G2_ST7565_LM6059_1_4W_SW_SPI
Can you also check the u8x8/flipmode example?

Thanks!

fcd, 2016-12-27:
To get the display to be "right side up" I had to make line 405 0x0a0 and line 406 0xc8

Flip mode 0 is upside down and 1 is right side up.  

I also had to change the default_x_offset to 1 (from 2) or I was getting some dots on the side of the screen.

olikraus, 2016-12-27:
ok, fine, i made the changes, but, what about flipmode_x_offset? Must this be set to 3?
More over, what about the flip  mode sequences. Shell i revert the flip mode sequence patterns? --> I did so in the latested snapshot.

I have checked in everything, also the latest release is here:
https://github.com/olikraus/U8g2_Arduino/archive/master.zip

Thanks again.

fcd, 2016-12-27:
Everything appears correct in your latest build.

olikraus, 2016-12-27:
oh, wow, thanks a lot

closed 2016-12-27 by olikraus