olikraus/u8g2 #124: HAL reset

iotexpert, 2016-12-25:
In the file u8x8_display.c in function void u8x8_d_helper_display_init(u8x8_t *u8x8)

You issue these commands:
      u8x8_gpio_SetReset(u8x8, 1);
      u8x8_gpio_Delay(u8x8, U8X8_MSG_DELAY_MILLI, u8x8->display_info->reset_pulse_width_ms);
      u8x8_gpio_SetReset(u8x8, 0);
      u8x8_gpio_Delay(u8x8, U8X8_MSG_DELAY_MILLI, u8x8->display_info->reset_pulse_width_ms);
      u8x8_gpio_SetReset(u8x8, 1);
      u8x8_gpio_Delay(u8x8, U8X8_MSG_DELAY_MILLI, u8x8->display_info->post_reset_wait_ms);

I think that you should make a new message "U8X8_MSG_RESET" and then add that code to the HAL.   I think that would be a little bit cleaner software architecture... and would handle the case of no HW reset a little bit cleaner (e.g. my case where there are only I2C lines)

olikraus, 2016-12-25:
Thanks a lot for such a deep analysis of the code.
I think the best would be to handle this msg as part of the gpio handler.

iotexpert, 2016-12-25:
Yes... that would improve the software architecture.  If you will let me Ill fix it and do a pull request.

olikraus, 2016-12-26:
No problem, i will work on this. I do not know exactly how to do this. Currently i am thinking of a combination of a reset message (handled ob byte level) and a reset pin handled on gpio level.

olikraus, 2016-12-26:
Sure...

closed 2017-05-20 by olikraus