olikraus/u8g2 #125: HAL in_transfer

iotexpert, 2016-12-26:
In a number of places there is a variable in_transfer which is set/unset.  The only reason I could think to do that is because in some situation the code might be re-entrant.  Is that right?

olikraus, 2016-12-26:
ok, probably a more difficult answer required here. I may also not have the time to give a complete answer as of now.
Short answer: No, the variable in_transfer has nothing to do with reentrant code. In fact the opposite is true, because in_transfer is a static local variable it will probably break the reentrant property of the rest of the code.

Let me discuss "reentrant" for a second. U8g2 has its own data structure, which u8x8_struct and u8g2_struct (a C++ like extension of u8x8_struct). So basically u8g2 is reentrant as long as individual structures are used. However this is not true for some low level activities like byte transfer. In fact it is conidered, that everything between the transfer start and transfer end is not interrupted. From that perspective the use of the static local variable should be fine. However, could would be more cleaner if in_transfer would be located in u8x8_struct itself. On the other side, it would be a waste of one byte if you are not require this in_transfer variable. In fact this variable is only used in u8x8_cad_uc16xx_i2c to implement some speedup in the byte protocoll of the i2c interface for the UC16xx family of display controllers.

So lets go one step back:

iotexpert, 2016-12-26:
When I looked at the code I thought that the variable was un-needed...e.g. not really used because you could never get there unless you were not in the middle of a transfer.  But I thought that you must have put it there for some good reason... which is why I asked the question.

olikraus, 2016-12-26:
Yes... reason is there.

closed 2017-05-20 by olikraus