olikraus/u8g2 #126: how to use chinese

coloz, 2016-12-26:
hi,Oliver
i'm a chinese,come from arduino chinese community (http://www.arduino.cn).
A long time ago I wrote a u8glib tutorial,
I am writing a tutorial about u8g2,Want to know how to display Chinese,
use the printUTF8 example,but chinese text shows a "=".
Hope to get your help

olikraus, 2016-12-26:
You have to assign a Chinese font first.

olikraus, 2016-12-26:
See here for some suitable fonts.
https://github.com/olikraus/u8g2/wiki/fntgrpunifont

No doubt, glyphs might be missing. Let me know which glyphs should be there. Also, if you have different font style...Let me know.

coloz, 2016-12-26:
thanks a lot,
Most of the Chinese fonts style need to give money, adobe produced a free sytle:
https://github.com/adobe-fonts/source-han-sans/

coloz, 2016-12-26:
Commonly used Chinese characters:
http://clz.me/u8g2/font_zh_3500.hex

coloz, 2016-12-26:
I use the arduino 101, the above 2MB flash chip, it can store more characters , may be more beneficial for users in China, Korea and Japan

olikraus, 2016-12-26:
Have you seen the PrintUTF8 example?
It should demonstrate the use of Chinese glyphs.

coloz, 2016-12-27:
I use it, but can not display Chinese,it shows:
hello world
=

olikraus, 2016-12-27:
Ohhhh, you got any errors? What is your board? Arduino Uno?

coloz, 2016-12-27:
My board is arduino 101,the display device is SSD1306 Through iic
Display English letters is normal, but unicode can not be displayed

olikraus, 2016-12-27:
I don't have access to my hardware at the moment. I will test this as soon as possible. Maybe you can post the code which does not work.

coloz, 2016-12-27:
Very strange,
I directly open example in IDE, and compile, not shown in Chinese
But after save as the example to other addresses, and then compile, is all right
I suspect this is an IDE bug.

thank you very much,your libs is great!

closed 2016-12-27 by coloz

sunlose, 2017-01-16:
Hi Olikraus, I had the same issue by copy the example of UTF8 from your post. I believe it may be Chinese character missing from your font list. For example, I tried "Hello World" or "你好世界” and screen can show it. But when I change to other Chinese character, nothing happens...

olikraus, 2017-01-16:
Indeed, this is a big problem. The font will bekome very large if i include all the chinese chars, so I decided to add only a subset. I spent some time to figure out what is a suitable subset, but for sure, there is a better subset possible.

Basically i have three different files with increasing number of chinese glyphs:

https://github.com/olikraus/u8g2/blob/master/tools/font/build/chinese1.map
https://github.com/olikraus/u8g2/blob/master/tools/font/build/chinese2.map
https://github.com/olikraus/u8g2/blob/master/tools/font/build/chinese3.map

All the numbers refer to the corrsponding unicode number.
If you use a _chinese1 font in u8g2, then only the chars from chinese1.map are included.

coloz, 2017-01-16:
@sunlose 使用 chinese3 可以显示更多的中文，但大部分Arduino都没有那么大的存储空间。推荐使用Arduino due和Arduino 101，这两个板子现在u8g2是支持的。
Use chinese3 can show more Chinese, but most of the Arduino are not so much storage space. Arduino due and Arduino 101 are recommended, and now u8g2 supports them

sunlose, 2017-01-17:
Thanks all. Let me try Chinese 3 to see.

发自我的 iPhone

disassembly, 2017-03-07:
when using u8g2.setFont(u8g2_font_unifont_t_chinese3);
I got this error: undefined reference to `u8g2_font_unifont_t_chinese3'
collect2.exe: error: ld returned 1 exit status
ide is 1.6.5

olikraus, 2017-03-07:
Due to the memory size, some very large fonts are disabled.

disassembly, 2017-03-07:
Thank you very much. 
Could a  nodemcu board have enough memory to try it?
Anyway, u8g2 is really a very powerful library, good work! 
I just used it to drive an unknown display with 192x64 px*px departed from a U key of a bank.

olikraus, 2017-03-07:
It would require to update line 167 here:
https://github.com/olikraus/u8g2/blob/master/csrc/u8g2.h

Will ESP8266 support arrays with more than 32K? If yes, i can enable large font support.

disassembly, 2017-03-08:
I delete line 167 and 169 in the code you mentioned and works fine on
nodemcu with an esp8266 12( maybe 12e, much the same).

oxp-edward, 2017-07-11:
how to generate chinese ttf file for u8g2?

coloz, 2017-07-11:
@oxp-edward tools下有个bdfconv工具，可以取模字体 https://clz.me/u8g2-bdfconv/

olikraus, 2017-07-11:
@oxp-edward First step is to create a bdf file out of the ttf file. This can be done with the following tool: 
https://www.math.nmsu.edu/~mleisher/Software/otf2bdf/
But also fontforge will worg.
Then use bdfconv as mentioned by @coloz

larryli, 2017-09-14:
花了一点时间，把文泉驿点阵宋体转了 https://github.com/larryli/u8g2_wqy

9pt 及 12x12 点阵完整的 GB2312 字符集 7539 个字符（含 ASCII）占用 208,228 字节。

disassembly, 2017-09-14:
太棒了，辛苦，谢谢！

RobotGB, 2017-12-17:
Hi, I've created a font file in .c using bdfconv so that I will be able to use my own font. So that I can verify it is valid, where do I place the file?
Kind regards,
RobotGB

olikraus, 2017-12-17:
You can copy the content of the generated c-file directly into your .ino code. The generated c code contains a c array. Use the array name as argument to setFont command.

RobotGB, 2017-12-18:
Thanks for the quick response.

RobotGB, 2017-12-18:
Hello again,
Upon pasting the code and compiling, I get this message:
`ccN8P4Dh.ltrans1.o:(.text+0x4): undefined reference to `u8g2_font_f16_t_japanese2'`
By the way, `u8g2_font_f16_t_japanese2` is the name of the font I am trying to use.
How can I change it to get it to work work?

Kind regards,
RobotGB

RobotGB, 2017-12-19:
Update: After deleting lines 167 and 169 (following disassembly's instructions), I get this message:
`C:\Users\RobotM0\Documents\Arduino\libraries\U8g2\src\clib\u8g2_fonts.c:160530:15: error: size of array 'u8g2_font_f16_t_japanese2' is too large`
Is there a way to aid this?

Kind regards,
RobotGB

olikraus, 2017-12-22:
This is a limitation of the compiler. I think the AVR gcc is limited here. You use ATMega for you project? I strongly suggest to use an ARM based Arduino Board (Zero or Due).

RobotGB, 2017-12-23:
Thanks
Kind regards,
RobotGB