olikraus/u8g2 #127: Slow HW SPI, as slow as I2C

gronat, 2016-12-26:
There are two 0.9'' SSD1306 OLEDs, one on I2C and the latter on 4w SPI and Arduino Nano V3. When I2C is used at 800kHz, displaying a text and an arrow (measured pressure value and gauge, not important) takes 55ms on average. When SPI is used to display the same screen it takes 50ms on average. Why is the HW SPI so slow?

I increased the the SPI from default 4MHz to 8MHz, indeed, I measured the CLK signal via oscilloscope and the measured peak to peak period was roughly 126ns. But to display the same screen it still takes 50ms. From this it follows that there must be a huge overhead between sending the chunks of data.

olikraus, 2016-12-26:
How can I rate this? I would need much more information. But 50ms sounds good for the nano.

gronat, 2016-12-26:
Hi olikarus,
8MHz SPI is as fast as 800kHz I2C? There is an order of magnitude higher clock speed, less overhead (no addressing, no ACK), while the performance is the same. That looks like a huge potential for improvement. It looks weird, I would expect SPI performance to be 3x 5x faster w.r.t. I2C.

I have only a low end DSO but it looks like that between each byte transfer there is a huge delay. I am willing to help with this issue. Can you point me tho the part of the code that handles transaction?

olikraus, 2016-12-26:
Code is inside U8x8lib.cpp.
I just use wire.h and SPI.h.
Can you post your U8g2 constructors please.

gronat, 2016-12-26:
Hi olikarus, Kindly find constructors below:
'
// OLED display I2C
U8G2_SSD1306_128X64_NONAME_1_HW_I2C u8g2(U8G2_R0);
// OLED display SPI
U8G2_SSD1306_128X64_NONAME_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
'
I compared the performance for both the default frequencies and the modified frequencies. The trend is the same, hence the same speed for I2C and SPI in terms of displaying identical data.

SPI freq changed 4000000UL -> 8000000UL in 
./src/clib/u8x8_d_ssd1306_128x64_noname.c line195
./src/clib/u8x8_d_ssd1306_128x64_noname.c line230
verified by oscilloscope at CLK pin

I2C freq changed in Twi.h from 400000 to 8000000
line 28:  #define TWI_FREQ 800000
verified by oscilloscope at SCK pin

I am using Arduino 1.6.12. It could be that the SPI library is written in sub-optimal way. If one checks the bus via oscilloscope it can be seen that most of the time the bus is idle between bytes. Sure, there is an overhead in filling the SPI register and handling the data but it should be in the order of dozens of cpu cycles, not the microseconds. My intuition is that SPI should run significantly (3x-5x) faster than IIC.

olikraus, 2016-12-27:
OK, I see. Can you try the F version of the constructor?
E.g. U8G2_SSD1306_128X64_NONAME_F_HW_I2C 

It should be faster because the overhead is lesser.

Moreover there should be a sequence of bytes which are transferred without delay in between.

gronat, 2016-12-27:
F version of the SPI takes roughly 32ms. Because of the display buffer, one has to deal with the memory issues. I am surprised that performance of the SPI is that bad. After finishing my project, I might deep dive into the code and try to figure out what is the source of the overhead. Thank you for your time and this awesome library.

olikraus, 2016-12-30:
Currently u8g2 just transfers one byte with SPI.transfer.
At least since 1.6.5 there is also:
SPI.transfer(buffer, size)
However buffer is overwritten with the incoming data.

olikraus, 2016-12-30:
I have checked the SPI.h code for AVR. I do not think that we can do much here....

closed 2016-12-30 by olikraus

gronat, 2017-01-21:
Now I see the problem too. I might fork the spi libraty to optimize it.

olikraus, 2017-01-21:
A multi byte function which ignores the incoming data would be good.