olikraus/u8g2 #129: stucks here !!  u8g > u8g2 ,.... help.... !!!!

svishnu06, 2016-12-29:
I am working with  SSD1306_128X64 OLED.  i switched to new library from ' u8g '  to 'u8g2' .   
in old library (u8g)  i used a line of code   "  **`u8g.print(String (percent) + "%");`** "  for display , one mapped value 'percent'  with '%"  symbol..   but  after switching to new u8g2,  i could not find any working code to  replace this .   help me please..

thank you...

olikraus, 2016-12-29:
Strange. I have not changed this part. Did you change IDE version also?
As a workaround: just print the % sign in a separate print statement.

svishnu06, 2016-12-29:
Yes . I changed  IDE.  Installed  new version  ,  now its working..  sorry for bothering you. .   Thank you for your support.

closed 2016-12-29 by svishnu06