olikraus/u8g2 #130: Can't use Wire with u8x8

reneverschoor, 2016-12-29:
While testing an I2C module I wanted to show some results on an OLED display instead of to the serial console.
That didn't work...
After some testing I reduced the code to:

#include <Arduino.h>
#include <SPI.h>
#include <Wire.h>
#include <U8x8lib.h>

U8X8_SSD1306_128X64_NONAME_SW_I2C u8x8(SCL, SDA, U8X8_PIN_NONE);

void setup(void)
{
  u8x8.begin();
  //Wire.begin()  // Add this, and the OLED will not siplay anything anymore :(
}

void loop(void)
{
  u8x8.setFont(u8x8_font_artossans8_r);
  u8x8.drawString(0, 0, "Hello World!");
}

The moment I add ``Wire.begin()`` nothing is displayed anymore.

olikraus, 2016-12-29:
Hi
Problem: Wire.h does not work with U8g2 SW_I2C (which uses digitalWrite instead of Wire.h)

PossibleSolutions:
1. Do not use Wire.begin(). Is it required?
2. Use U8g2 HW_I2C (which uses Wire.h instead)
3. Use different pins for the OLED (replace SCL and SDA with some other pin numbers)

reneverschoor, 2016-12-29:
Hi,

1. ``Wire.begin()`` is required, otherwise I can't use any of the Wire methods to talk to my I2C module.
2.  That works! Using ``U8X8_SSD1306_128X64_NONAME_HW_I2C u8x8(U8X8_PIN_NONE);`` did the trick.
3. Good tip, I hadn't realized that I could just use other pins than SCL/SDA. Also works.

Thanks for the quick reply & solution.

olikraus, 2016-12-29:
no problem

closed 2016-12-29 by olikraus