olikraus/u8g2 #131: in program screen rotation

soheils2, 2016-12-29:
hey there 
first i wanna thank u for this awesome library.

in u8glib (v1) we can change screen rotation in  the code for even each draw with this code:

u8g.setRot180();

but in this version you can just set the rotation on  the begining and there is no way to rotate it in the program ,
OR i couldnt find the void that do this. 

please let me now if there is a way to change the rotation in program. tnx

olikraus, 2016-12-29:
There is no function as such. Maybe I add this..

  u8g2->cb = rotation;
  u8g2->cb->update(u8g2);

olikraus, 2016-12-30:
oh, it is already there:
```
   void setDisplayRotation(const u8g2_cb_t *u8g2_cb) {
      u8g2_SetDisplayRotation(&u8g2, u8g2_cb); }
```
But indeed the documentation is missing.

olikraus, 2016-12-30:
documentation done

closed 2016-12-30 by olikraus