olikraus/u8g2 #136: Lanmu SSH1106 - DrawBitmap

Neilski, 2017-01-01:
I have a _LANMU 1.3" SSH1106 SPI I2C IIC 128X64 OLED LCD LED Display Module Board_ that I am trying to use with with the U82G library.  Everything works as expected (`.drawStr()`, `.drawCircle()` etc.) apart from when I try `.drawBitmap()`  which appears to draw the each column of the image, but in the wrong order.

I have used the following U8G2 constructor `U8G2_SH1106_128X64_NONAME_1_4W_SW_SPI`, but I have also tried different constructor configurations.   I have also tried using the  [U8G2 logo](https://github.com/olikraus/u8g2/blob/master/doc/u8g2_logo_128x64.xbm) but this seemed to suffer the same problem.  Here is my code...

```
#include <U8g2lib.h>

static unsigned char bitmap[] = {
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xe0, 0x00, 0x00, 0x00, 0x00, 0xf8,
   0x01, 0x00, 0x00, 0x00, 0xfe, 0x01, 0x00, 0x00, 0x00, 0xff, 0x01, 0x00,
   0x00, 0x80, 0xff, 0x01, 0x00, 0x00, 0xc0, 0xff, 0x01, 0x00, 0x00, 0xe0,
   0xff, 0x00, 0x00, 0x00, 0xf0, 0xff, 0x00, 0x00, 0x00, 0xf0, 0xfb, 0x00,
   0x00, 0x00, 0xf8, 0xf9, 0x00, 0x00, 0x00, 0xf8, 0xf8, 0x00, 0x00, 0x00,
   0xfc, 0xf0, 0x01, 0x00, 0x00, 0x7c, 0xf0, 0x01, 0x00, 0x00, 0x7c, 0xf0,
   0x01, 0x00, 0x00, 0x7e, 0xf0, 0x03, 0x00, 0x00, 0x3e, 0xe0, 0x03, 0x00,
   0x00, 0x3e, 0xe0, 0x07, 0x00, 0x00, 0x3e, 0xc0, 0x0f, 0x00, 0x00, 0x3e,
   0xc0, 0x1f, 0x00, 0x00, 0x3e, 0x80, 0x3f, 0x00, 0x00, 0x3e, 0x00, 0x7f,
   0x00, 0x00, 0x3e, 0x00, 0xff, 0x01, 0x00, 0x7e, 0x00, 0xfe, 0x0f, 0x3e,
   0x7c, 0x00, 0xf8, 0xff, 0x7f, 0x7c, 0x00, 0xf0, 0xff, 0x7f, 0xfc, 0x00,
   0xc0, 0xff, 0x7f, 0xf8, 0x00, 0x00, 0xff, 0x3f, 0xf8, 0x01, 0x00, 0xf0,
   0x3f, 0xf0, 0x03, 0x00, 0x80, 0x1f, 0xf0, 0x07, 0x00, 0xc0, 0x1f, 0xe0,
   0x0f, 0x00, 0xe0, 0x0f, 0xc0, 0x3f, 0x00, 0xf8, 0x07, 0x80, 0xff, 0x01,
   0xff, 0x03, 0x00, 0xff, 0xff, 0xff, 0x01, 0x00, 0xfe, 0xff, 0xff, 0x00,
   0x00, 0xf8, 0xff, 0x3f, 0x00, 0x00, 0xe0, 0xff, 0x0f, 0x00, 0x00, 0x00,
   0xff, 0x01, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
};

/* U8G2 Constructor */
// U8G2_SH1106_128X64_NONAME_F_4W_SW_SPI u8g2(
U8G2_SH1106_128X64_NONAME_1_4W_SW_SPI u8g2(
  U8G2_R0,  // rotation
  D5,       // clock
  D7,       // data
  D8,       // cs
  D2,       // dc
  D1        // reset
);

void setup(void) {
  u8g2.begin();
}

void loop(void) {
  u8g2.firstPage();
  do {
    u8g2.drawBitmap(44, 12, 5, 40, bitmap);
    // u8g2.drawCircle(64, 32, 20, U8G2_DRAW_ALL);
  } while ( u8g2.nextPage() );

  // u8g2.clearBuffer();
  // u8g2.drawBitmap(44, 12, 5, 40, bitmap);
  // u8g2.sendBuffer();  
}
```
[image]

I should say I am new to this environment so I suspect it is more likely to be my mistake than a problem with the library!  Apologies if this is the incorrect forum.

olikraus, 2017-01-01:
How is the bitmap generated?
If there is already a gif file, use gimp and save the gif as XBM file (which is actually c code).
Use the XBM file content together with drawXBM.

Neilski, 2017-01-01:
Thank you.  The data was already generated as XBM, so changing the method call from `.drawBitmap()` to `.drawXBM()` was all that was required.  I has misunderstood the use of `.drawBitmap()` - my apologies.

closed 2017-01-01 by Neilski