olikraus/u8g2 #140: add st7628  or any colored nokia lcd support?

soheils2, 2017-01-03:
hey 
i just working with my new project and i want to know if u can add this lcd to your library?

soheils2, 2017-01-03:
https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=1&cad=rja&uact=8&ved=0ahUKEwijtvzRsKbRAhVIKywKHdEABgoQFggZMAA&url=https%3A%2F%2Fgithub.com%2Fkr4fty%2FST7628-Nokia-1600-LCD-Library&usg=AFQjCNHFi6qBK2pXeVUMIhh1MH-3gLs3kg&sig2=rGqEtjWhmr9TFZQIlDhPMA&bvm=bv.142059868,d.bGg

olikraus, 2017-01-03:
u8g2 is a library for monochrome displays.

soheils2, 2017-01-03:
okay but why? in last version there is some color lcds in supported devices

https://www.youtube.com/watch?v=PyO0zI5jXAc

[image]

this is what i writing with your library and some other libs ; for now i used monochrome  lcd that will be awesome if u can add colored lcds to the lib again...

olikraus, 2017-01-03:
Nice work.

For color lcds/oleds: i have split the code into ucglib (color) and u8g2 (monochrome).
However: I can not add a new device without having it physically on my desk.

Blue-Marlin, 2017-01-04:
There is a fundamental difference between:
* Colour displays - where you can set any pixels colour at your will
* Coloured displays - like in the picture above - where some pixels (upper quarter?) have an other colour than the others.

closed 2017-01-08 by olikraus