olikraus/u8g2 #141: add thickness to drawline

soheils2, 2017-01-03:
in feature versions u can add thickness to the lines  
OR 
add angle into the darwbox(); void  

for now i draw thicker lines by drawing lots of lines together and change the xy , it can be a built-in "void" for thicker lines

PS; sorry about my bad english knowledge :("

olikraus, 2017-01-03:
Either adding thickness or adding angle into drawbox() will make the library very large. Also algorithms are very complicated.
I suggest to use drawTriangle() 
At least adding two triangles would make up one "Quadrilateral" 
You could also do the required math for the "rotated box" by yourself and use drawTriangle.

Conclusion: 
1. Lines with thickness and rotated boxes is very complicated. At least i will not add this, but i would be happy to accept any pull requests.
2. I think the problem can be solved by using drawTriangle with some additional math operations.

closed 2017-01-08 by olikraus

drfailov, 2024-03-16:
To achieve thickness effect I made this calculations to triangulate desired form by 2 triangles. My tests showed than there's no gap between triangles.
Hust replace `lcd()->` by your u8g2 call.

```
void drawThickLine(float centerX, float centerY, float toX, float toY, float thickness){ 
  float dx = toX-centerX;
  float dy = toY-centerY;
  float d = sqrt(dx*dx+dy*dy);
  float ux = dx/d;
  float uy = dy/d;
  float mux = ux*thickness;
  float muy = uy*thickness;
  float rx1 = muy;   //https://limnu.com/sketch-easy-90-degree-rotate-vectors/
  float ry1 = -mux;
  float rx2 = -muy;
  float ry2 = mux;

  float x0 = centerX+rx1;
  float y0 = centerY+ry1;
  float x1 = centerX+rx2;
  float y1 = centerY+ry2;
  float x2 = toX+rx1;
  float y2 = toY+ry1;
  float x3 = toX+rx2;
  float y3 = toY+ry2;
  lcd()->drawTriangle(/*x0*/x0, /*y0*/y0, /*x1*/x1, /*y1*/y1, /*x2*/x2, /*y2*/y2);
  lcd()->drawTriangle(/*x0*/x1, /*y0*/y1, /*x1*/x3, /*y1*/y3, /*x2*/x2, /*y2*/y2);
}
```