olikraus/u8g2 #148: U8G2_PCD8544_84X48 and U8X8_PCD8544_84X48

dreep, 2017-01-06:
Hello,

First, thx for your work !!!

1. So, i don't know if i'm good.
When i use that code (based on  "Hello world" U8G2lib) :
`U8G2_PCD8544_84X48_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 23, /* data=*/ 25, /* cs=*/ 29, /* dc=*/ 27, /* reset=*/ 31);  // Nokia 5110 Display

// End of constructor list

void setup(void) {
  u8g2.begin();
}

void loop(void) {
  u8g2.clearBuffer();					// clear the internal memory
  u8g2.setFont(u8g2_font_5x7_tr);	// choose a suitable font
  u8g2.drawStr(0,20,"Hello World!");	// write something to the internal memory
  u8g2.sendBuffer();					// transfer internal memory to the display
  //delay(1000);
  u8g2.drawFrame(80,0,4,48);
  u8g2.sendBuffer();  
  delay(1000);
}`
i obtain a screen when i can see "Hello world !", it's ok, but drawFrame which draw an empty rectangle, i see something like that :
                             _
                             ||
                           ||
                         ||
                       ||
                             -
Sorry for that, but it's what i see...
I did something wrong ?

2. If i want to use u8x8 library, can i modify the number of row and column ? Because on a little screen like Nokia 5110, i have a tile 10x6... it's a little small and i have lot's of place on the screen between each letter...

Thx a lot and sorry for my poor english... Normal french people, you know... :D

olikraus, 2017-01-06:
Nice Ascii art, but still, can you send a picture? The issue is not clear to me. What exactly is wrong?

The u8x8 makes use of the internal memory architecture of the display to avoid any local RAM usage. With this restriction, only 8x8 tiles (chars) are possible. If you want to use something else, you have to use u8g2.

dreep, 2017-01-06:
Lol !

[image]

olikraus, 2017-01-06:
oh

olikraus, 2017-01-06:
meanwhile i have reproduced this problem on my own display..

dreep, 2017-01-06:
Si i'm not crazy or incompetent...

olikraus, 2017-01-06:
not at all... bugs are always possbile... This one happend at an unexpected place.

You can download the fixed version here:
Install via Arduino IDE (Add libraray zip file)

dreep, 2017-01-06:
Another question if it's possible...
I have lcdmenulib with Arduino méga, my screen and a 2 axes joystick with button liké that : https://i1.wp.com/henrysbench.capnfatz.com/wp-content/uploads/2015/09/Arduino-PS2-Joystick-Logo.png
With my i2c liquidcrystal, i used the switch without problem on pin 22...
Now, i can't... 
Can i have an interaction between your lib and my switch ?

dreep, 2017-01-06:
Thx a lot for fix bug !

olikraus, 2017-01-06:
> Now, i can't...
What exactly does this mean?
U8g2 only uses those pins which are given to the constructor. 
However, pin 22 and 23 are very close. Maybe there is a crostalk from pin  22 to pin 23.

dreep, 2017-01-06:
I will try another pin...

dreep, 2017-01-06:
Ok, it's good for my first question... so, bug is closed i think.
Very impressed for fast reply and work !
Thx a lot.

olikraus, 2017-01-06:
no problem.

closed 2017-01-06 by olikraus