olikraus/u8g2 #149: SSD1306 128x64 I2C display offset

ros87, 2017-01-06:
Arduino UNO with a noname ssd1306 128x64 I2C running the u8x8 helloworld example sketch.

U8X8_SSD1306_128X64_NONAME_SW_I2C u8x8(/* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE);

Display is offset -2 pixels to the left.

Any way to change the offset?

[image]

olikraus, 2017-01-06:
This is a SH1106 OLED... Hmm.. I should add this to my FAQ list.

ros87, 2017-01-06:
Ooh, if so I'm sorry. Seller listed it as SSD1306. I'll change the config and retry tomorrow.

olikraus, 2017-01-06:
I have updated my draft version of the FAQ:
https://github.com/olikraus/u8g2/blob/master/doc/faq.txt

Let me know whether the SH1106 constructor works.

ros87, 2017-01-06:
Ok thank you.

I see no SH1106 I2C constructor in the library (2.10.1)

ros87, 2017-01-07:
Sorry, found it and tried:

U8X8_SH1106_128X64_NONAME_SW_I2C u8x8(2,0,U8X8_PIN_NONE);

No display at all.

olikraus, 2017-01-07:
What about this?
U8X8_SH1106_128X64_NONAME_SW_I2C u8x8(/* clock=/ SCL, / data=/ SDA, / reset=*/ U8X8_PIN_NONE);

ros87, 2017-01-07:
[image]

closed 2017-01-07 by ros87

olikraus, 2017-01-08:
Thanks for the feedback :-)

omegalotti, 2019-03-26:
for the record, I may also have the same oled 128x64 screen. Because mine also has VCC first, GND as the second in.
My version worked flawlessly with this one:
U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);

I also have a 128x32 display but the pin layout is GND+VCC+SCL+SDA 
It worked with U8G2_SSD1306_128X32_UNIVISION_F_HW_I2C u8g2(U8G2_R0);

Thank you Oli for such a great library and faq!

olikraus, 2019-03-26:
:-)

bjornstevens, 2020-09-28:
> 
> 
> What about this?
> I mean, in the original constructor, just replace SSD1306 with SH1106

This fixed it all for me!

Icyrockton, 2021-04-28:
very useful 
**U8G2_SSD1306_128X64_NONAME_F_SW_I2C**

**U8G2_SH1106_128X64_NONAME_F_SW_I2C**