olikraus/u8g2 #152: U8G2 List example and control

dreep, 2017-01-10:
Hello,

Can we use SelectionList example with joystick (like [http://henrysbench.capnfatz.com/henrys-bench/arduino-sensors-and-input/arduino-ps2-joystick-tutorial-keyes-ky-023-deek-robot/](url)) ?

Thx a lot

olikraus, 2017-01-10:
Hi
I added the selection list and other menu option just as a simple example. I think it should be a user task to add suitable user controll input methods. In fact it will beyond the scope of u8g2 to implement menu system. Goal is to support monochrome displays.
So please treat the selection list as an example. Of course you can reuse the code, etc.

dreep, 2017-01-10:
ok, thx, for answer !!!
questions close !

olikraus, 2017-01-10:
I think this is very similar to the issue #150. It would make up a complete new library to support all kind of input methods.

closed 2017-01-15 by olikraus