olikraus/u8g2 #153: IST3020

olikraus, 2017-01-10:


olikraus, 2017-01-10:
IST3020_ERC19264
192x64
http://www.buydisplay.com/default/4-3-inch-cog-192x64-display-graphic-lcds-module-ist3020-black-on-white

olikraus, 2017-01-10:
please download the new version from here:
https://github.com/olikraus/U8g2_Arduino/archive/master.zip

Add the zip lib file in the library manager of the Arduino IDE.

Uncomment and update the constructor, execute the u8x8/FlipMode example. Will it work?

DefekC, 2017-01-11:
First of all, thanks a lot!! :)
Secondly, it works but with few fixes.
Tested with ERC19264-1 [balck-on-white](http://www.buydisplay.com/default/4-3-inch-cog-192x64-display-graphic-lcds-module-ist3020-black-on-white) & [white-on-black](http://www.buydisplay.com/default/4-3-inch-192x64-graphic-cog-lcd-module-display-ist3020-white-on-black) using 6800-series (parallel) communication.

**Fixes:**
- internal oscillator initialization missing in init sequence after first reset:
`static const uint8_t u8x8_d_ist3020_erc19264_init_seq[] = {`
`U8X8_START_TRANSFER(),             	/* enable chip, delay is part of the transfer start */`
`U8X8_C(0x0e2),            			/* soft reset */`
`U8X8_C(0x0ab),            			/* FIX: start internal oscillator */`
`...`
`}`

- corrections to default flipmode:
`U8X8_C(0x0a0),		                /* FIX: ADC set to non-reverse */`
`U8X8_C(0x0c8),		                /* FIX: COM set to reverse */`

- default V0 voltage resistor ratio & contrast value changed:
`U8X8_C(0x020),		                /* FIX: v0 voltage resistor ratio, default was 0x026*/`
`U8X8_CA(0x081, 0x019),			/* FIX: set contrast, contrast value, default was 0x027*/`

- changes in basic display info values:
`static const u8x8_display_info_t u8x8_ist3020_erc19264_display_info =`
`{`
`...`
`/* tile_width = */ 24,			/* FIX: width of 24*8=192 pixel */`
`...`
`/* pixel_width = */ 192,`
`...`
`}`
};

I already noticed that there are other minor details to be changed, but this is the minimum fix-kit to make display light up (see image of results [here](http://i63.tinypic.com/2j5ysts.jpg) and [here](http://i63.tinypic.com/125tant.jpg)).

olikraus, 2017-01-11:
Thanks a lot for your valuable input.
If have updated the repo and you can download the fixed version from here:
https://github.com/olikraus/U8g2_Arduino/archive/master.zip

Can you test this again?
Again, please tell me whether FlipMode.ino example works.

Thanks!

DefekC, 2017-01-12:
Tested! And works with minor changes!
**Fixes:**

- default LDC bias on initialization (_forgot to mention previously_):
`U8X8_C(0x0a3),		                /* FIX: LCD bias 1/7, old value was 1/9 (0x0a2) */`

- display info: flipmode x ofset:
`/* flipmode_x_offset = */ 64, /* old value was 0 */`

There are inconsistencies [in IST3020 datasheet](http://www.buydisplay.com/download/ic/IST3020.pdf) that does not match reality... well, at least in case of this particular display(s) from [buydisplay.com](http://www.buydisplay.com). Driver supports **resolution up to 256x64, not 224X65 as stated in datasheet**. Also voltages (V0 and supply voltages) has values that does not match datasheet, but as I doubt anyone would ever use external power sources instead of built-in regulators, this is not an issue.

olikraus, 2017-01-12:
Again, thanks a lot for the precise input.
The new version is here:
https://github.com/olikraus/U8g2_Arduino/archive/master.zip

I have also updated the timing according to the IST3020 datasheet.

olikraus, 2017-01-15:
ok, i assume it will work now... so i close this issue.

closed 2017-01-15 by olikraus