olikraus/u8g2 #156: ST7920 shows garbage @ 20MHz

trashmaster9000, 2017-01-14:
hi,

in my current project i use an ATMega328 standalone at 20MHz.
for easier programming i have the same Display connectetd to a Arduino nano.

on the Arduino nano my program works fine but on the ATmega it shows strange lines and dots.

a other programm (written with u8glib) works fine at 20Mhz.

The display is connected via SW SPI, on both controllers on the same pins.

[image]

olikraus, 2017-01-14:
Which constructor is this?

trashmaster9000, 2017-01-14:
U8G2_ST7920_128X64_F_SW_SPI u8g(U8G2_R1, 6, 7, 8);

olikraus, 2017-01-14:
I have slowed down speed a little bit.
Please download U8g2 from here:
https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Install via Arduino IDE, Add lib zip file...

trashmaster9000, 2017-01-14:
ok, it works now.
thanks ;)

closed 2017-01-14 by trashmaster9000

olikraus, 2017-01-14:
wow, fast response. thanks