olikraus/u8g2 #158: HW i2c on ESP8266 Arduino

sticilface, 2017-01-15:
Hi. 

I'm using your lib with great success but a little confused about a few things. 

My display is the `SH1106`

and i have it working using this constructor
```cpp
U8G2_SH1106_128X64_NONAME_F_SW_I2C display(U8G2_R0, /* clock=*/ 4, /* data=*/ 5, /* reset=*/ U8X8_PIN_NONE);   // All Boards without Reset of the Display
```

however,  the Arduino esp8266 has wire.h available an implemented with speeds up to 400Khz. Currently a screen redraw takes 130ms, and i belief that u8g2 is bit banging this out with emulated i2c rather than using the wire.h.  is this correct?

sticilface, 2017-01-15:
OK.  I've figured out what is going on.  

I'm just trying using the HW_I2C constructor, but i've got SDA and SDL mixed up on my PCB such that clock is pin4, and data is pin5.  The default Wire.begin() has them the other way round.  On the ESP8266 it is possible to use Wire.begin(SDA, SDL) 
```cpp
void begin(int sda, int scl);
```
so that these pins can be set dynamically.  Would you be able to alter your constructors so that HW_I2C at least on the ESP8266 can be called with the SDA, SDL specified.  

This drops a screen redraw from 140ms to 4ms in my sketch.  a massive difference!

olikraus, 2017-01-15:
Hmmm... i could extend the constructor
U8G2_SSD1306_128X64_NONAME_F_HW_I2C(const u8g2_cb_t *rotation, uint8_t reset = U8X8_PIN_NONE, uint8_t sda = U8X8_PIN_NONE, uint8_t scl = U8X8_PIN_NONE)
In such a case the SDA and SCL pins could be provided for the ESP8266, but could be ignored for others.

Then, the procedure 
u8x8_SetPin_HW_I2C(getU8x8(), reset);
has to be extended to
u8x8_SetPin_HW_I2C(getU8x8(), reset, sda, scl);

 * [ ] Update u8x8_SetPin_HW_I2C in U8x8lib.cpp
 * [ ] Update u8x8_byte_arduino_hw_i2c in U8x8lib.cpp
 * [ ] Update codebuild.c

iotexpert, 2017-01-15:
This is the spot where the HAL should be fixed rather than expanded...

olikraus, 2017-01-15:
I do not think that the HAL is affected here. This is just the Arduino specifc part of a specific hardware implementation. 
But maybe you can be more specific what exactly we should do instead.

sticilface, 2017-01-15:
That would be awesome!  👍

olikraus, 2017-02-05:
code update 
https://github.com/olikraus/u8g2/commit/8975938ca0e1a05723d771140ba6d7c1ff411103

olikraus, 2017-02-05:
test successful

closed 2017-02-05 by olikraus

sticilface, 2017-02-13:
checked back in to find you've made progress.  But i can't test just yet..  work... but I will.  thanks

olikraus, 2017-02-13:
Great. I did only some basic tests with my own board. Any feedback is welcome.

sticilface, 2017-02-14:
So I've tried to use the lib, but it has not worked out so far.  Maybe you can help.

I can compile the demo, using the following 
```
U8G2_SSD1306_128X32_UNIVISION_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE, /* clock=*/ SDA, /* data=*/ SCL);   // pin remapping with ESP8266 HW I2C
```

however the screen that I am using is 64 pixels high, and there does not seem to be a version for that. 

additionally I was using this constructor perfectly before
```
U8G2_SH1106_128X64_NONAME_F_HW_I2C display(U8G2_R0);
```

as reading the issues it is the one to use to avoid that missing first column of pixels.  

So is it possible to have a version of SH1106, with HW I2C pin remapping, for 128x64 screen sizes?

sticilface, 2017-02-14:
my bad...
```
U8G2_SH1106_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, U8X8_PIN_NONE, 4, 5);
```

olikraus, 2017-02-14:
:-)  Pin remapping is there with all I2C constructors...