olikraus/u8g2 #159: Font and RAM usage

sticilface, 2017-01-15:
Hi

Sorry for another post.  I've looked through the docs and issues and can't seem to find an answer to this question.  

Fonts seem to be held in RAM.  Removing reference to 
```
display.setFont(u8g2_font_logisoso30_tf);
```

increased my available RAM by 7Kb.  

I've got two questions:

1) Is there a possibility of storing these fonts in Flash is available.  The ESP8266 which is what I use, has abundant flash storage, and precious RAM!  

2)  are there scalable fonts, that can be used and resized without needed to include each size separately?

many thanks

A

olikraus, 2017-01-15:
I just found this:
http://tech.scargill.net/accessing-esp8266-arrays-in-flash/

All in all this looks complicated.

sticilface, 2017-01-15:
maybe...  but then maybe not..  here is an example sketch that stores a massive array in flash, and retrieves it!  

```cpp

ICACHE_RODATA_ATTR uint8_t test[6429]  = {
  0,22,35,252,247,15,3,223,8,17,32,255,251,21,251,16,
  251,0,0,0,15,0,1,4,16,16,15,5,0,96,240,240,
  240,240,240,240,96,96,96,96,96,0,96,96,96,7,7,7,
  15,3,9,238,238,238,238,68,68,68,12,19,38,15,1,254,
  12,192,12,192,12,192,12,192,12,192,12,192,127,240,127,240,
  25,128,25,128,25,128,255,224,255,224,51,0,51,0,51,0,
  51,0,51,0,51,0,10,20,40,15,2,253,12,0,12,0,
  61,128,127,128,195,128,193,128,192,0,224,0,126,0,31,128,
  1,192,0,192,192,192,225,192,255,128,223,0,12,0,12,0,
  12,0,12,0,11,15,30,15,2,0,60,0,102,0,195,0,
  195,0,102,0,60,0,1,192,15,0,56,0,231,128,12,192,
  24,96,24,96,12,192,7,128,11,14,28,15,2,0,30,0,
  63,0,99,0,96,0,96,0,48,0,56,0,124,192,111,192,
  199,128,195,0,199,128,255,224,124,224,3,7,7,15,5,8,
  224,224,224,224,224,64,64,5,20,20,15,6,252,24,56,48,
  96,96,96,192,192,192,192,192,192,192,192,96,96,96,48,56,
  24,5,20,20,15,3,252,192,224,96,48,48,48,24,24,24,
  24,24,24,24,24,48,48,48,96,224,192,10,11,22,15,2,
  5,12,0,12,0,12,0,204,192,255,192,63,0,12,0,63,
  0,51,0,115,128,97,128,12,12,24,15,1,1,6,0,6,
  0,6,0,6,0,6,0,255,240,255,240,6,0,6,0,6,
  0,6,0,6,0,5,6,6,15,4,253,56,56,112,96,192,
  128,11,2,4,15,2,6,255,224,255,224,3,3,3,15,5,
  0,224,224,224,11,20,40,15,2,253,0,96,0,96,0,192,
  0,192,1,128,1,128,3,0,3,0,6,0,6,0,12,0,
  12,0,24,0,24,0,48,0,48,0,96,0,96,0,192,0,
  192,0,10,16,32,15,2,0,30,0,127,128,97,128,192,192,
  192,192,192,192,192,192,192,192,192,192,192,192,192,192,192,192,
  192,192,97,128,127,128,30,0,10,16,32,15,2,0,28,0,
  252,0,252,0,12,0,12,0,12,0,12,0,12,0,12,0,
  12,0,12,0,12,0,12,0,12,0,255,192,255,192,11,16,
  32,15,1,0,31,128,63,192,112,224,96,96,96,96,0,96,
  0,224,1,192,3,128,7,0,14,0,28,0,56,0,112,0,
  255,224,255,224,11,16,32,15,1,0,31,128,63,192,112,224,
  96,96,0,96,0,224,1,192,15,128,15,192,0,224,0,96,
  0,96,192,96,224,224,127,192,63,128,11,16,32,15,1,0,
  3,128,7,128,15,128,13,128,25,128,25,128,49,128,49,128,
  97,128,97,128,255,224,255,224,1,128,1,128,15,224,15,224,
  11,16,32,15,2,0,127,192,127,192,96,0,96,0,96,0,
  111,0,127,192,113,192,0,224,0,96,0,96,0,96,192,224,
  225,192,127,192,63,0,10,16,32,15,2,0,7,192,31,192,
  60,0,112,0,96,0,224,0,223,0,255,128,225,192,192,192,
  192,192,192,192,224,192,113,192,127,128,31,0,10,16,32,15,
  2,0,255,192,255,192,192,192,0,192,1,128,1,128,1,128,
  3,0,3,0,3,0,6,0,6,0,6,0,12,0,12,0,
  12,0,10,16,32,15,2,0,30,0,127,128,225,192,192,192,
  192,192,192,192,97,128,63,0,127,128,225,192,192,192,192,192,
  192,192,225,192,127,128,63,0,10,16,32,15,3,0,30,0,
  127,128,97,128,192,192,192,192,192,192,192,192,193,192,227,192,
  126,192,60,192,0,192,1,128,3,128,255,0,252,0,3,11,
  11,15,5,0,224,224,224,0,0,0,0,0,224,224,224,5,
  14,14,15,3,253,56,56,56,0,0,0,0,0,56,56,112,
  96,192,128,13,12,24,15,1,1,0,56,0,240,3,192,15,
  0,60,0,240,0,240,0,60,0,15,0,3,192,0,240,0,
  56,12,6,12,15,1,4,255,240,255,240,0,0,0,0,255,
  240,255,240,13,12,24,15,1,1,224,0,120,0,30,0,7,
  128,1,224,0,120,0,120,1,224,7,128,30,0,120,0,224,
  0,9,15,30,15,3,0,126,0,255,0,195,128,193,128,193,
  128,1,128,3,128,15,0,28,0,24,0,24,0,0,0,24,
  0,24,0,24,0,10,18,36,15,2,254,28,0,127,0,99,
  0,193,128,193,128,199,128,207,128,221,128,217,128,217,128,221,
  128,207,192,199,192,192,0,192,0,97,128,127,128,30,0,14,
  15,30,15,0,0,63,0,63,0,7,128,7,128,12,192,12,
  192,28,224,24,96,24,96,63,240,63,240,112,56,96,24,252,
  252,252,252,13,15,30,15,1,0,255,192,255,224,48,112,48,
[... 336 lines cut ...]
  0,6,0,14,0,12,0,28,0,254,0,254,0};

uint8 read_rom_uint8(const uint8* addr){
    uint32 bytes;
    bytes = *(uint32*)((uint32)addr & ~3);
    return ((uint8*)&bytes)[(uint32)addr & 3];
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);

  Serial.printf("\n\nBegin\n"); 
  for (int i = 0; i < 6429; i++) {
    Serial.printf("%u,",  read_rom_uint8( &test[i]) );
    if (i % 50 == 0) { Serial.println(); }
  }
   Serial.printf("\n\nend\n"); 
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

sticilface, 2017-01-15:
Of course I've not idea how the array is used within a font, or how to maybe implement it alongside other frameworks.  The esp8266 does support PROGMEM as well, which might be an option.

sticilface, 2017-01-17:
I've thought I'd have a stab. adjusting what you created...

```
#if defined(__GNUC__) && defined(ESP8266)
#  include "c_types.h"
#  define U8X8_FONT_SECTION(name) ICACHE_FLASH_ATTR
#  define u8x8_pgm_read(adr) pgm_read_byte_near_esp(adr)
#  define U8X8_PROGMEM PROGMEM
#endif
```

However, just including this gives the ```section `.irom0.text' will not fit in region `irom0_0_seg'``` error.  without including any fonts.  Not sure why that is happening as using ICACHE_FLASH_ATTR should just put it in flash!  and the macro above should just expand to that.. `ICACHE_FLASH_ATTR` which is itself `__attribute__((section(".irom0.text")))` so I'm not sure why this message is occurring. 

I defined `pgm_read_byte_near_esp` as 

```cpp
uint8_t pgm_read_byte_near_esp(uint8_t * addr) {
    uint32_t bytes;
    bytes = *(uint32_t*)((uint32_t)addr & ~3);
    return ((uint8_t*)&bytes)[(uint32_t)addr & 3];
}
```

sticilface, 2017-01-17:
Ok, so i think that it does work, as it try to link and include every font that is included!  it compiles if i just include a few fonts.  

```
40236190 g     O .irom0.text	000037f3 u8g2_font_osb41_tf
```

and `pgm_read_byte_near_esp` works just fine.  So that is most of the battle.  Now just need to work out why it tries to use unlinked fonts!

olikraus, 2017-01-17:
ok, maybe this helps: https://github.com/esp8266/Arduino/issues/734

sticilface, 2017-01-17:
That makes sense.  I've always failed to fiddle with the linkers though. The fixes these seem to suggest that all functions within .c files should now got to FLASH by default and need RAM attribute to go to ram! However, the font array's go by default to ram not flash and I'm not sure how to implement the objcopy stuff!!

olikraus, 2017-01-17:
I will have a look at this, but i can also not promise anything.

olikraus, 2017-01-18:
hmmm... both option look a little bit strange to me
1. How can this work? Shell wir force the user to add the font twice: Include statement and font name in the code?
2. I do not see a difference to the current situation.

Anyhow, i guess i need to investigate this by myself....

olikraus, 2017-01-18:
i did some tests. conclusion:
Arrays have to be declared as
static const unsigned char array[64] ICACHE_RODATA_ATTR = { ... };
garbage collections works as usual
probably this access function is required.

olikraus, 2017-01-18:
current page buffer HelloWorld.ino looks like this:

kraus@one2linux:/tmp/arduino_build_28208$ /home/kraus/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-size HelloWorld.ino.elf 
   text	   data	    bss	    dec	    hex	filename
 225905	   4568	  29776	 260249	  3f899	HelloWorld.ino.elf

olikraus, 2017-01-18:
After my fixes, it looks like this:
```
kraus@one2linux:/tmp/arduino_build_28208$ /home/kraus/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-size HelloWorld.ino.elf
   text	   data	    bss	    dec	    hex	filename
 228209	   2440	  29776	 260425	  3f949	HelloWorld.ino.elf
```

olikraus, 2017-01-18:
with several other fonts, the size is this
```
kraus@one2linux:/tmp/arduino_build_28208$ /home/kraus/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-size HelloWorld.ino.elf
   text	   data	    bss	    dec	    hex	filename
 253377	   2440	  29776	 285593	  45b99	HelloWorld.ino.elf
```
Looks good....
Maybe someone can do another test. 2.13.0 is here:
https://github.com/olikraus/U8g2_Arduino/archive/master.zip

sticilface, 2017-01-19:
This looks to work to me.  Good job.  I'ma bit confused though as your changes are fairly similar to ones i tried but it didn't work, and u didn't change fonts to this ```static const unsigned char array[64] ICACHE_RODATA_ATTR = { ... };``` that i can see from your commits.. so how does this work? just curious

olikraus, 2017-01-19:
Actually, most of the information was there already from you. I did setup my measure tool (size calculation) so that i can measure the results. Then I created an example code and i started to measure what all is required to avoid the RAM copy of an array.
After that phase, conclusion was this:
1) The array must be declared as "const" ("static" does not matter)
2) The array must be ICACHE_RODATA_ATTR, which is nothing else than some irom section.
 
Of course knowing, that i have to make 32bit aligned reads did help a lot. Thanks for this.

In the second phase i modified u8g2, however it did not work. Yet it was a different error message. Something about "can not fit into section". So i started to analyse the linker script... still without much result (i only noticed, that the array can be placed in .text also). Finally i remembered an issue, which i had once with the AVR gcc: Each array has to be in a spearate section for the garbage collection, so i again updated the macros and took over the concept from AVR. Each font is now placed in its own section call .text.<fontname>
This will 
- Place the font in FLASH only (no RAM usage)
- Allow the linker garbage collector to remove unused fonts.

So finally, we can use the same concept as in AVR: Just use the font in your code. No extra includes required. :-)

sticilface, 2017-01-19:
ah!  see i removed the separate name sections when i did it as i thought that was something specific to AVR.. very clever.  Works like a treat and now u can have a lot... of fonts with no ram usage at all.

olikraus, 2017-01-19:
no doubt, this was a nice finding. I did not expect, that the ESP8266 has similar issues like AVR. On ARM systems the problem does not seem to exist.

closed 2017-01-19 by olikraus