olikraus/u8g2 #160: ESP8266 Arduino SSD1606

SouDGitHub, 2017-01-15:
Hi guy,

In case you want to update your small table with compatibility, I can tell that this EINK screen controller SSD1606 is working with the ESP8266 / Arduino.
U8G2_SSD1606_172X72_F_4W_HW_SPI u8g2(U8G2_R0, EINK_CS, EINK_DC, EINK_RST);

Here is my pinout (EINK BS1 pin connected to GND)  :
//int EINK_MOSI = D7;
//int EINK_SCLK = D5; 
int EINK_CS = D8; 
int EINK_DC = D3;
int EINK_RST = D2;

The good things is that I'm still able to use I2C peripherals using 
int SDAPin = D9;
int SCLPin = D10;
Wire.begin(SDAPin,SCLPin);

But the problem is that the begin function refreshes the screen that is not always wanted using deep sleep mode :(
Will try in an other way on my sketch!

Thank you for your library. It is fine for beginners like me !

Nico.

olikraus, 2017-01-15:
Thanks for sharing this information. I am happy that u8g2 works for you.

> But the problem is that the begin function refreshes the screen that is not always wanted using deep sleep mode :(

You do not need to call u8g2.begin() at the beginning. You could call u8g2.begin() just before you want to write something to the eInk display.

SouDGitHub, 2017-01-17:
Yes it is what I've done.
But I use this screen to display a small calendar and the temperature that I refresh every 10 minutes using the deep sleep mode. The refreshing sequence could have been less intrusive in my code but anyway I will EASILY deal with this !

Thank you for your library, and please do not understand my message as a real issue!
BTW, I don't know about others constructors like 3Wire SPI mode and others... Maybe someone else will try it to save pins in this ESP-12E board !

Nico.

olikraus, 2017-01-17:
:-)

SouDGitHub, 2017-01-17:
Using the same pinout as describe post#1 , the SW 4W SPI constructor works fine
U8G2_SSD1606_172X72_F_4W_SW_SPI(rotation, clock, data, cs, dc [, reset])
U8G2_SSD1606_172X72_F_4W_SW_SPI u8g2(U8G2_R0, EINK_SCLK, EINK_MOSI, EINK_CS, EINK_DC, EINK_RST);

the SW 3W SPI constructor doesn't work.
U8G2_SSD1606_172X72_F_3W_SW_SPI(rotation, clock, data, cs [, reset])
U8G2_SSD1606_172X72_F_3W_SW_SPI u8g2(U8G2_R0, EINK_SCLK, EINK_MOSI, EINK_CS, EINK_RST);

The first methods releases the MISO pin to be used for something else !

Will not go further. Have fun :)

closed 2017-01-28 by olikraus

ric2b, 2017-02-04:
Sorry for commenting on a closed issue but is there a way use deep sleep without flashing the screen (turns off and then back on immediately) on reset?

> You do not need to call u8g2.begin() at the beginning. You could call u8g2.begin() just before you want to write something to the eInk display.

I did try to call begin() right before writing to the screen but it still flashes. I also tried to call initDisplay() + setPowerSave(0) instead, same result. Is there something I can change to avoid this behaviour?

olikraus, 2017-02-04:
Are you refering to SSD1606 ePaper devices?
Assuming so, the answer is this:
 1. There is no setPowerSave support for SSD1606
 2. The fashing is part of the init sequence.  

To my current knowledge, flashing seems to be part of the pixel setting, but actually my knowledge is very limited on these devices and whatever I have implemented is just taken from the datasheets. And according to the datasheets, the flashing seems to be required.

In fact documentation is very poor here.

ric2b, 2017-02-05:
Ah, I must've misread when I commented, mine is an SSD1306 64*48 OLED.
Is it possible to avoid flashing those on and off when coming back from deep sleep?

Thank you and sorry for commenting on an unrelated issue :/

olikraus, 2017-02-05:
no problem, but i have to investigate this. 
New issue created --> #180