olikraus/u8g2 #162: SSD1309 constructor does not work

olikraus, 2017-01-16:
copyied from #113 

> I'm just testing this, but get this compile error:
> Error compiling for board Arduino/Genuino Uno
> Build failed for project 'ISU'
> ccO8oWXd.ltrans1.ltrans.o*: In function global constructors keyed to 65535_0_ISU.cpp.o.9315
> ccO8oWXd.ltrans1.o*: (.text.startup+0xaa): undefined reference to u8g2_Setup_ssd1309_128x64_noname_1
> 
> collect2.exe*: error: ld returned 1 exit status
> 
> with constructor:
> U8G2_SSD1309_128X64_NONAME_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=/ 2, / data=/ 3, / cs=/ 4, / dc=/ 5, / reset=*/ U8X8_PIN_NONE);
> 
> The error does not mean much to me, any ideas?
> Using VMicro in VS2015 community

OP uses 2.12.6, Lib was fully delted first.

olikraus, 2017-01-16:
BTW, for me it did work nicly:
https://github.com/olikraus/u8g2/wiki/gallery#13-jan-2017-ssd1309-128x64

hoss183, 2017-01-16:
Ok its working now. Possibly that i had 2 copies of library, or possibly that i needed to reload toolchain on VS. 
solution was to delete all/both libraries, delete in VS, reinstall by Arduino IDE, then rescan toolchain in VS.
Thanks

olikraus, 2017-01-16:
:-)
ok, great.

closed 2017-01-16 by olikraus