olikraus/u8g2 #163: ST75256

olikraus, 2017-01-17:
JLX256128

olikraus, 2017-02-15:
cad 011

the ST75256 has the same I2C protocol as the SSD13xx, BUT: for arguments have the data bit set.
this means, the ssd13xx i2c cad proc can not be used because this is a cad 001 procedure, but here we need a cad 011 procedure.

olikraus, 2017-02-18:
Current tests are done with u8x8:

```
U8X8_ST75256_JLX256128_4W_SW_SPI u8x8(/* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);	
//U8X8_ST75256_JLX256128_4W_HW_SPI u8x8(/* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);	
```

[image]

Status: The display which I have received seems to be brocken. 2 from 6 columns are not visible (4G mode). Monochrome mode does not work at all.

7134956, 2017-03-04:
It worked for me

[u8x8_d_st7586s_jlx240160g666.c.txt](https://github.com/olikraus/u8g2/files/818677/u8x8_d_st7586s_jlx240160g666.c.txt)
[u8x8_d_st75256_jlx240160g676.c.txt](https://github.com/olikraus/u8g2/files/818678/u8x8_d_st75256_jlx240160g676.c.txt)

olikraus, 2017-03-04:
So my display is broken?

7134956, 2017-03-04:
These functions I wrote some time ago. Your code I have not tested. Look at my code ...
[image]

olikraus, 2017-03-09:
I have tested your ST75256 code: no output at all.

olikraus, 2017-09-16:
20PIN SPI COG 25664 LCD Module ST75256 : I2C Interface, 5V
VDD=5V?
ST75256_256x64LCD_pinout.jpg

olikraus, 2017-10-11:
status: flip mode seems to be possible, see c code. Implementation for both displays can be done. The scan dir command not only changes the auto update direction, but also modifies the scan direction.

closed 2017-10-12 by olikraus