olikraus/u8g2 #165: SSD1322 256x64

MsiSoftwareEngineer, 2017-01-18:
Olikraus, thank you so much for such an amazing library! 
We are currently using U8glib for the SSD1322 256x64 display. Constructor: U8GLIB_NHD31OLED_GR

Is there plans for this Controller to be ported over? We are using a US Micro P21301 for our device and can provide testing feedback if needed.

Thanks,

olikraus, 2017-01-18:
This is done already:
https://github.com/olikraus/u8g2/wiki/gallery#2-oct-2016-ssd1322-256x64-nhd-display

`U8G2_SSD1322_NHD_256X64_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);`

Remember to enable U8G2_16BIT in u8g2.h

Oliver

uriba107, 2017-01-23:
Adding to this question,
in the past I've been using the BW constructor on u8glib to gain some performance boost (call it wishful thinking). Is there a similar option in u8g2? or is it always GR/BW?

Thanks,
Uri

olikraus, 2017-01-23:
In U8g2 there is only BW.
It was a design decision during the development of the second version of u8glib not to support graylevel.

uriba107, 2017-01-23:
Thanks for clearing that up.

Keep up the fantastic job :)

closed 2017-01-28 by olikraus