olikraus/u8g2 #166: u8g2.firstPage(); - Why does it loop?

davixete, 2017-01-19:
Hi oliver

I have a doubt. I have written a basic example.

`#include <Arduino.h>
#include <U8g2lib.h>

// ARDUINO MEGA GLCD
U8G2_KS0108_128X64_1 u8g2 (U8G2_R0, 22,23,24,25,26,27,28,29, /* enable = */ 37, /* dc = */ 36, /* cs0 = */ 33, /* cs1 = */ 34, /* cs2 = */ U8X8_PIN_NONE, /* reset = */ U8X8_PIN_NONE);   // Set R/W to low!

const int rwPin = 35;

void setup() {
   Serial.begin(9600);
   u8g2.begin();
   digitalWrite(rwPin, LOW); // ponemos el pin r/w a LOW para el GLCD

   u8g2.firstPage();
    do {
      Serial.println("PASADAS"); 
    } while ( u8g2.nextPage() );

}

void loop() {
  // put your main code here, to run repeatedly:

}`

Why is it repeated 6 times?

thank you very much

olikraus, 2017-01-19:
hehe, i was always wondering why noone asked this before.
But it is actually an excellent question. In U8g2 i took over the concept which i introduced in U8glib.
So please have a look there:
https://github.com/olikraus/u8glib/wiki/tpictureloop

davixete, 2017-01-19:
Now I understand the number of repetitions :).
I used this method to save RAM.
In the manual it says that the full buffer requires more RAM.
I'll try the full buffer.
I feel my English from google translate.
Many thanks oliver. Awesome job!.

olikraus, 2017-01-19:
Thanks...
At the end, you have to decide: Faster, but more RAM consumption, slower with lesser RAM consumption. U8g2 now offers both. ;-)

davixete, 2017-01-21:
Problem solved. Thanks again oliver.

closed 2017-01-21 by davixete