olikraus/u8g2 #167: ST7920 - Real Time System

lebertagna, 2017-01-20:
I'm using a 128x64 Display based on ST7920, the U8g2lib and a 8080 protocol to make a real time system, but I need the refresh rate to be faster. 

What do I have to do to make it faster?

[image]

olikraus, 2017-01-20:
Use full buffer HW_SPI with Arduino Zero.

lebertagna, 2017-01-23:
It didn't work. 
I use the conecction: CS/RS = 3, reset = 4 and PSB = gnd.

What am I doing wrong?

olikraus, 2017-01-23:
hmmm what about full buffer 8080 mode?

lebertagna, 2017-01-23:
full buffer 8080 mode is what I'm using

olikraus, 2017-01-23:
well then... you didn't wrote this... but anyway, it will be difficult to improve speed here...

viti95, 2017-01-26:
It's possible to obtain ~15 fps using an ST7920 (full buffer HW SPI, also full buffer 8080) with an Arduino Mega, and ~25-30 fps with an Arduino Due. I can show you an example made by me (Assetto Corsa + Arduino Mega)

https://www.youtube.com/watch?v=sGgVgdKVZ4s

olikraus, 2017-01-26:
Maybe you should post your code. Then i might be able to make some suggestions for speed improvements.

olikraus, 2017-01-28:
no further input... closing this. Feel free to reopen this.

closed 2017-01-28 by olikraus

lwFlyFish, 2018-03-07:
I have a bit of a problem, can I share your code?