olikraus/u8g2 #169: T6963 constructor/wiring

JamesGKent, 2017-01-23:
i have a display that uses the T6963 controller that i am looking to use, but having difficulty working out how to wire it up, i know that how the pins are labeled will vary by manufacturer, but i notice that the constructor for using thos display seems to have changed how many pins it expects from u8glib to u8g2:  
the ug8lib constructor:  
`U8GLIB_T6963_240X128(d0, d1, d2, d3, d4, d5, d6, d7, cs, a0, wr, rd [, reset])`

and the u8g2 constructor:
`U8G2_T6963_240X128_1_8080(rotation, d0, d1, d2, d3, d4, d5, d6, d7, enable, cs, dc [, reset])`

where the new constructor seems to expect one less pin?
the screen has:  
chip enable pin (assuming this is `cs` on old, but which on new?)
one labeled as C/D which i believe is `a0` on the old and `dc` on the new constructors?
but i am unsure of what to do with the `wr` and `rd` pins from the screen with the new constructor?

olikraus, 2017-01-23:
Yes, i cleaned up the constructor. In the examles you will find this:
//U8G2_T6963_240X128_1_8080 u8g2(U8G2_R0, 8, 9, 10, 11, 4, 5, 6, 7, /*enable=*/ 17, /*cs=*/ 14, /*dc=*/ 15, /*reset=*/ 16); // Connect RD with +5V, FS0 and FS1 with GND

I added a comment here, saying, that RD input must be connected to 5V. The previous constructor just made the same, but i think it is better not to waste a pin here just to pull one line to 5V.

> chip enable pin (assuming this is cs on old, but which on new?)

The "cs" pin has not been changed. This is the same name. However I changed the sequence of the pins.

> one labeled as C/D which i believe is a0 on the old and dc on the new constructors?

Yes, to be more consistent i decided to use dc (data/command) everywhere in U8g2, avoiding the other names. These labels are all the same: CD = DC = A0 = RS

I have updated my FAQ a little bit:
https://github.com/olikraus/u8g2/blob/master/doc/faq.txt

My plan is to create a proper wiki page from that FAQ, but at the moment i am just collecting questions.

JamesGKent, 2017-01-23:
thank you, this explains what i needed, I've been trying (via trial and error) to get this display working but so far haven't managed to get anywhere... I'll try this tonight and see how it goes.
Thanks
James

closed 2017-01-28 by olikraus