olikraus/u8g2 #171: Feature suggestion: gauges

Giuse1977, 2017-01-25:
First of all I'd like to thank you for your libraries!
I'd like to share a simple code for drawing basic gauges using your library as an example .
It would be nice to have a similar function integrated in your library!

``
/*Gauges 0.1  25 JAN 17
 * Giuseppe Bellomo  bellomo.giuseppeATgmail.com
 * Draws simple gauge based on a value passed to the function.  
 *   USAGE
 *   Drawgauge(x,y,r,p,v,minVal,maxVal,t )
 *   Draws a gauge with center in x,y (x is an int since u8g2lib supports screen with a width of more than 255, y is a byte to keep variable footprint as low as possible); 
 *   with radius r; 
 *   needle lenght is p percent of radius; 
 *   representing value v(must be an int); 
 *   minVal and maxVal are the min and max values of v (integer); 
 *   t is gauge type: 
 *      0 for quarter left,
 *      1 for quarter right, anticlockwise
 *      2 for upper half semicricle
 *      3 for 3/4 starting at -180
 *      4 for 3/4 starting at -90
 *      very easy to add other versions
 */

#include <U8g2lib.h> //Include u8g2 library
U8G2_ST7920_128X64_1_HW_SPI u8g2(U8G2_R0,10); //my display is connected this way

float gs_rad; //stores angle from where to start in radinats
float ge_rad; //stores angle where to stop in radinats

//example values for testing, use the values you wish to pass as argument while calling the function
byte cx=50; //x center
byte cy=20; //y center
byte radius=20; //radius
byte percent=80; //needle percent

void setup() {
u8g2.begin();
}

void loop() { //this loop is for testing gauges mode
  for (byte t=0; t <=4; t++){
  
    for (int c=0; c <= 100; c++) {  //this loop is for testing needle movement
      Drawgauge(cx,cy,radius,percent,c,0,100,t);
    }
  }
}

void Drawgauge(int x, byte y, byte r, byte p, int v, int minVal, int maxVal,  byte t ) {
int n=(r/100.00)*p; // calculate needle percent lenght

u8g2.firstPage();
      do { 
          switch (t){
            case 0: { //left quarter
                float gs_rad=-1.572; //-90 degrees in radiant
                float ge_rad=0;
                float i=((v-minVal)*(ge_rad-gs_rad)/(maxVal-minVal)+gs_rad);
                int xp = x+(sin(i) * n);
                int yp = y-(cos(i) * n);
                   u8g2.drawCircle(x,y,r, U8G2_DRAW_UPPER_LEFT );
                   u8g2.drawLine(x,y,xp,yp); 
                }
                break;
            case 1: { //right quarter, needle anticlockwise
          
                float gs_rad=0;
                float ge_rad=1.572; //90 degrees in radiant
                float i=((v-minVal)*(ge_rad-gs_rad)/(maxVal-minVal)+gs_rad);
                int xp = x+(cos(i) * n);
                int yp = y-(sin(i) * n);
                   u8g2.drawCircle(x,y,r, U8G2_DRAW_UPPER_RIGHT );
                   u8g2.drawLine(x,y,xp,yp); 
                }
                break; 
            case 2: { //upper half
                float gs_rad=-1.572;
                float ge_rad=1.572;
                float i=((v-minVal)*(ge_rad-gs_rad)/(maxVal-minVal)+gs_rad);
                int xp = x+(sin(i) * n);
                int yp = y-(cos(i) * n);
                   u8g2.drawCircle(x,y,r, U8G2_DRAW_UPPER_LEFT|U8G2_DRAW_UPPER_RIGHT );
                   u8g2.drawLine(x,y,xp,yp); 
               }
               break;
            case 3: { //three quarter starting at -180
                float gs_rad=-3.142;
                float ge_rad=1.572;
                float i=((v-minVal)*(ge_rad-gs_rad)/(maxVal-minVal)+gs_rad);
                int xp = x+(sin(i) * n);
                int yp = y-(cos(i) * n);
                   u8g2.drawCircle(x,y,r, U8G2_DRAW_UPPER_LEFT|U8G2_DRAW_UPPER_RIGHT|U8G2_DRAW_LOWER_LEFT ); 
                   u8g2.drawLine(x,y,xp,yp); 
                } 
                break;
            case 4: { //three quarter starting at -90
                float gs_rad=-1.572;
                float ge_rad=3.141;
                float i=((v-minVal)*(ge_rad-gs_rad)/(maxVal-minVal)+gs_rad);
                int xp = x+(sin(i) * n);
                int yp = y-(cos(i) * n);
                  u8g2.drawCircle(x,y,r, U8G2_DRAW_UPPER_LEFT|U8G2_DRAW_UPPER_RIGHT|U8G2_DRAW_LOWER_RIGHT );
                  u8g2.drawLine(x,y,xp,yp); 
                }
                break;
          }
  
 } while ( u8g2.nextPage() );
   
}
``

olikraus, 2017-01-25:
Thanks for sharing this code.
I am a little bit unsure to add this to the core part of the library. However i would be glad to add this as an example code.
Can you provide a full working .ino file. ok... it might be complete already, but maybe you can attach something like "Gauge.ino" to this issue.

Giuse1977, 2017-01-25:
Ciao Oliver
It's a working example, at least on my setup. Hope this example meets the format requirement to be included as an example, feel free to modify it to meet the standards.
Actually it only draws a very basic gauge, but I'm working on more complex and better looking ones for my project, and I will share improvements in future.
Thank you for your attention
Regards,

Giuseppe

[Gauge.zip](https://github.com/olikraus/u8g2/files/729927/Gauge.zip)

olikraus, 2017-02-06:
Hi Giuseppe
Can we put the code under the 2-clause BSD license (same license as u8g2)?
Oliver

Giuse1977, 2017-02-06:
Sure, you can! When I've time I'll try to improve or to make a function of it.

Thanks 

Giuseppe

olikraus, 2017-05-14:
added to the repo, but maybe some rework is required. 

Code is here:
https://github.com/olikraus/u8g2/tree/master/sys/arduino/u8g2_page_buffer/Gauge

closed 2017-05-14 by olikraus

Giuse1977, 2017-05-14:
Thanks for publishing my code. Please let me know what kind of rework may be required and I 'll try to do it :-)

Godfear17, 2018-08-28:
This is ridiculously useful! Thanks a ton for writing this!

Giuse1977, 2018-08-28:
Happy that is useful to someone ; )