olikraus/u8g2 #172: Can it be made smaller ?

AndKe, 2017-01-27:
While the library is impressive, the simple graphics test using U8X8_SSD1306_128X64_NONAME_SW_I2C u8x8
compiles to 9032 bytes or  32% of a 328

olikraus, 2017-01-27:
... which includes the huge Arduino Print and Wire libs

olikraus, 2017-01-27:
I have compiled U8x8 Helloworld for the Uno with Arduino 1.8.0. I get this:
U8X8_SSD1306_128X64_NONAME_SW_I2C 5660 Bytes (17%)
U8X8_SSD1306_128X64_NONAME_HW_I2C 5696 Bytes (17%)
U8X8_SSD1306_128X64_NONAME_4W_SW_SPI 5538 Bytes (17%)
U8X8_SSD1306_128X64_NONAME_4W_HW_SPI 5460 Bytes (16%)

So one first question is: Why does your result deviate that much?

olikraus, 2017-01-27:
Another interesting point is, that the Wire (TWI) lib is always included. It should be only included with U8X8_SSD1306_128X64_NONAME_HW_I2C, but is there also with all other constructors.

olikraus, 2017-01-27:
U8X8_SSD1306_128X64_NONAME_4W_SW_SPI could be reduced to 4500 if the inclusion of Wire lib. Of course not an option for your, becasue you need the wire interface.

AndKe, 2017-01-27:
I was referring to the u8x8/GraphicsTest  example,  compiling HelloWorld with 
```
U8X8_SSD1306_128X64_NONAME_SW_I2C u8x8(/* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE); 
```
for Pro Mini results in 5672 Bytes.  (so it's not that much off) 

your observation about TWI is essencial.

olikraus, 2017-01-27:
ok... not sure what i can do here. Can we close this issue?

AndKe, 2017-01-27:
ok, I thought SPI may been included when I2C was , and vice versa.  , but if you see nothing to improve , to make it smaller - I'll close the issuemyself.
Thank you for your time.

closed 2017-01-27 by AndKe