olikraus/u8g2 #173: ERC-13265 display support

nmarevic, 2017-01-28:
First of all, thank you very much for this great library.

Is it possible to add support for this display 
http://www.buydisplay.com/default/2-5-inch-132x65-arduino-lcd-display-module-serial-spi-black-on-white

all datasheets are available on that site.

Also, the controller SPLC502 in that display seems compatible with ST7565, but please correct me if I'm wrong.

Cheers

olikraus, 2017-01-28:
I have extended the command compare table here:
https://github.com/olikraus/u8g2/blob/master/doc/controller_cmds.txt

The SPLC502 indeed includes all ST7565 commands.

Can you do test with the following constructors?
U8X8_ST7567_PI_132X64_4W_SW_SPI
U8X8_ST7565_LM6059_4W_SW_SPI
U8X8_ST7565_EA_DOGM128_4W_SW_SPI 
Which one is best? I will take this as a starting point for the new device.
Feel free to test with other controllers also (see the controller_cmds.txt for other controller names)

olikraus, 2017-01-28:
if possible please send a picture of the display. Thanks.

nmarevic, 2017-01-29:
The display is on the way. I will post my results as soon as I receive it. Thanks

joaqrus, 2017-02-02:
I have problems with a very similar display East Rising ERC12864-1 Series. My display is exactly ERC12864FS-1.

It works but the pages/buffer are wrong. Y axe pixels start at the middle of the display (32).

You can see a photo and my configuration in the Arduino forum display: [Arduino Forum](http://forum.arduino.cc/index.php?topic=452310.0)

joaqrus, 2017-02-02:
The best fit is the constructor `U8X8_ST7565_LM6059_4W_SW_SPI`, the pixel quality  is really good but, as you see in the Arduino photo I attached, the Y axis is wrong.

olikraus, 2017-02-03:
@joaqrus: I have created a new topic #177 for the ERC12864FS-1

olikraus, 2017-04-09:
closing this, to clean up the issue list, feel free to reopen...

closed 2017-04-09 by olikraus