olikraus/u8g2 #175: U8X8 lib on AVR ATmega

prulitarit, 2017-01-31:
Hello! I'm trying to get u8x8 lib working in AVR Studio 4, but I can not find any documentation for AVR ATmega. Is u8g8 lib compatible with WinAVR and AVR Studio?

olikraus, 2017-01-31:
Basically it should work with the C interface. Just include all c and h files from this directory into your project:
https://github.com/olikraus/u8g2/tree/master/csrc

Use setup procedures from here:
https://github.com/olikraus/u8g2/wiki/u8g2setupc

There is also an example project for ARM:
https://github.com/olikraus/lpc11u3x-gps-logger/blob/master/u8g2_logo/src/main.c

prulitarit, 2017-01-31:
I'm not really programmer, but I can get u8glib working following your wiki's manual. u8glib works well, I only wanted get smaller code size with u8x8, because only textdraw procedure is used. But thank you for your work and support!

olikraus, 2017-01-31:
Basically it will be identical to u8glib, except that you have to use different sources.

closed 2017-02-05 by olikraus

olikraus, 2018-03-26:
you could post some code and try to explain your problem

maopiccoto, 2018-03-27:
Hello @olikraus , after a lot of reading and coffee I think I'm ready to answer to myself. I used code from different issues/users and is working! As I said in my earlier comment, I'm using ATmega328p and a NOKIA 5110 (this uses a PCD8544 driver I guess) in AtmelStudio7. 

During this coding I faced two issues I didn't know how to solve:
 1.`case U8X8_MSG_GPIO_AND_DELAY_INIT:`  I think this is the reason why my display sometimes seems to be dying (see attached picture) 
2. `_delay_ms(arg_int);` seems to return an error --> `_builtin_avr_delay_cycles expects a compile time integer constant` but another user worked out very easy.

[image]

All in all, this code is working pretty well right now and I leave it here to users looking for a hints to implement u8g2 for ATmega328p in C environment..

``

	#define F_CPU 8000000UL
	#include <avr/io.h>
	#include <util/delay.h>
	#include "u8g2.h"
	#include "u8x8.h"

	// PD4=4 SCLK 
	// PD5=5 MOSI 
	// PD7=7 DC 
	// PB0=8 CS 
	// PC0=14 RESET 14
    // PB5 =LED

	u8g2_t u8g2;

	uint8_t u8g2_gpio_and_delay_pcd8544(U8X8_UNUSED u8x8_t *u8x8, U8X8_UNUSED uint8_t msg, U8X8_UNUSED uint8_t arg_int, U8X8_UNUSED void *arg_ptr){
		
		switch(msg){
			//Initialize SPI peripheral
			case U8X8_MSG_GPIO_AND_DELAY_INIT:
				break;
			//Function which implements a delay, arg_int contains the amount of ms
			case U8X8_MSG_DELAY_MILLI:
				//_delay_ms(arg_int);
				for(uint8_t i=arg_int; i > 0; i--) { // Modified here, AVR5
					_delay_ms(1);
				}
				
				break;
			case U8X8_MSG_DELAY_10MICRO:
				_delay_us(10);
				break;
			//Function to define the logic level of the clockline
			case U8X8_MSG_GPIO_SPI_CLOCK:
				if (arg_int == 0) {				// arg_int=0: Output low at SCLK pin
					DDRD |= _BV(DDD4);
					PORTD &= ~(_BV(PORTD4));
				} else {						// arg_int=1: Input dir with pullup high for SCLK pin
					DDRD &= ~(_BV(PORTD4));		// The same with the rest of pins
					PORTD |= _BV(PORTD4);
				}
				break;
			case U8X8_MSG_GPIO_SPI_DATA:
				if (arg_int == 0) {
					DDRD |= _BV(DDD5);
					PORTD &= ~(_BV(PORTD5));
				} else {	
					DDRD &= ~(_BV(PORTD5));
					PORTD |= _BV(PORTD5);
				}	
				break;	
			case U8X8_MSG_GPIO_CS:	
				if (arg_int == 0) {
					DDRB |= _BV(DDB0);
					PORTB &= ~(_BV(PORTB0));
				} else {	
					DDRB &= ~(_BV(PORTB0));
					PORTB |= _BV(PORTB0);
				}	
				break;		
			case U8X8_MSG_GPIO_DC:		
				if (arg_int == 0) {
					DDRD |= _BV(DDD7);
					PORTD &= ~(_BV(PORTD7));
				} else {	
					DDRD &= ~(_BV(PORTD7));
					PORTD |= _BV(PORTD7);
				}
				break;		
			case U8X8_MSG_GPIO_RESET:
				if (arg_int == 0) {
					DDRC |= _BV(DDC0);
					PORTC &= ~(_BV(PORTC0));
				} else {	
					DDRC &= ~(_BV(PORTC0));
					PORTC |= _BV(PORTC0);
				}
				break;
			default:
				return 0; //A message was received which is not implemented, return 0 to indicate an error
			}							
		return 1; // command processed successfully.
	}

	/***************************************************************************

	***************************************************************************/

	void displayInit(void){
		u8g2_Setup_pcd8544_84x48_f(&u8g2, U8G2_R0, u8x8_byte_4wire_sw_spi, u8g2_gpio_and_delay_pcd8544);
		u8g2_InitDisplay(&u8g2);		//Send initialization code to the display
		u8g2_SetPowerSave(&u8g2, 0);
		u8g2_SetFont(&u8g2, u8g2_font_miranda_nbp_tr);
	}

	int main(void)
	{
		DDRB |= (1<<DDB5);
		displayInit();
		
		while (1)
		{
			u8g2_ClearBuffer(&u8g2);
			u8g2_SetFontMode(&u8g2,1);
			u8g2_SetDrawColor(&u8g2,1);
			u8g2_DrawRBox(&u8g2,0,12,82,15,3);
			u8g2_SetDrawColor(&u8g2,2);
			u8g2_DrawStr(&u8g2, 2,22, "u8g3 is awesome!!!");
			u8g2_SendBuffer(&u8g2);
			PORTB ^=(1<<PINB5);
			_delay_ms(1000);
		}
	}

So, What can I do to solve this blinky LCD? In fact I use u8g2 in Arduino and it's awesome, the digits are presented in a very clear way but I don't know how to solve this in C.

Finally, is there any improvement I should do to this code? Thanks for your help Oli.

Edit: At the end of this writing the LCD got full contrast again, so I'm not sure if this is due to my lack of coding skills or missing contacts in my board.

olikraus, 2018-03-27:
Amazing work. Thanks for your contribution.

I have one doubt: What do the commands DDRB |= (1<<DDB5); and PORTB ^=(1<<PINB5); do?
It DDRB |= (1<<DDB5); is kind of a setup (port direction setup), then you could better place this to the  U8X8_MSG_GPIO_AND_DELAY_INIT message.

maopiccoto, 2018-03-27:
That's Port configuration for a blinky LED for testing purposes. Maybe you can not see it in the picture but it's there.

logiclayer, 2020-01-04:
Hello olikraus,
I made work some samples about your librar in Arduino . Now I would like to try it first Atmel Studio and Atmega328P and then STM32F103C8T6. But in Atmel Studio I could not understand how to build construction. I'd like to use it in HardWare SPI mode. My LCD controller is ST7920.
Can you build a construction method for me ?

olikraus, 2020-01-05:
Did you check https://github.com/olikraus/u8g2/wiki/u8g2as7 ?

Note: The C++ API (with constructors) is only possible on Arduino. For any other development environments you can only use the C API (as you can see at the beginning of the thread)