olikraus/u8g2 #176: No other I2C traffic possible

AndKe, 2017-01-31:
Using 
U8X8_SSD1306_128X64_NONAME_SW_I2C u8x8(/* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE);  on Arduino Nano.

when using :
```
u8x8.begin(); 
```
..accessing another I2C device crashes it.

While this (regardless of order):
```
  Wire.begin();  
  u8x8.begin();
```
stops display from showing anything (but the application runs)

- I am using I2C like this: 

```
 
  Wire.beginTransmission(deviceAddress);
  Wire.write(func);
  Wire.endTransmission(false);
  
  Wire.requestFrom(deviceAddress,2,true);
   while(Wire.available() 
  b1 = Wire.read();
...

 ```

olikraus, 2017-01-31:
Please use
U8X8_SSD1306_128X64_NONAME_HW_I2C u8x8(reset=*/ U8X8_PIN_NONE);

AndKe, 2017-01-31:
oops , my bad :)

closed 2017-01-31 by AndKe

olikraus, 2017-01-31:
no problem, i should publish the FAQ more visible in the wiki.