olikraus/u8g2 #177: Support for ERC12864FS-1

olikraus, 2017-02-03:
moved from #173

Constructor U8X8_ST7565_LM6059_4W_SW_SPI: 
[Arduino Forum](http://forum.arduino.cc/index.php?topic=452310.0)

olikraus, 2017-02-03:
Please locate file u8x8_d_st7565.c on your local drive (maybe here: .../Arduino/libraries/U8g2_Arduino/src)

Reference: 
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_st7565.c

In line 412 you will see this command:
U8X8_C(0x060), /* set display start line to ... */

Please change the value to
U8X8_C(0x040), /* set display start line to ... */

Will this fix the issue?

Oliver

joaqrus, 2017-02-03:
Yes, it works after this small change!
[http://i.imgur.com/FvobGkr.jpg](url)

Thank you!

joaqrus, 2017-02-03:
By the way there are some fake pixels at the very right of the display. something similar hapened to me using your old library (uglib) using another display.

olikraus, 2017-02-03:
hmm locks like the display is shifted by one pixel to the right. Compare with this display:
https://raw.githubusercontent.com/wiki/olikraus/u8g2/img/st7567_pax_instruments.jpg

Please locate the following lines
  /* default_x_offset = */ 1,	/* not sure... */
/* flipmode_x_offset = */ 3,
in https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_st7565.c 
(should be somewhere in line 454). Please change this to
  /* default_x_offset = */ 2,
/* flipmode_x_offset = */ 2,

Will this fix the problem?

joaqrus, 2017-02-03:
Vielen Danken Oliver!, spitzes Support!, Das Problem is gelöscht.

I followed your instructions, changing the X-axis pixels offset definition (from 1/3 to 2/2), problem was solved, the pixels are now off. However I  put the former values back and, interesting enough, the pixels are still off...

joaqrus, 2017-02-03:
This is how it looks like now:
[http://i.imgur.com/loH9Sym.jpg](url)

joaqrus, 2017-02-03:
[Imgur](http://i.imgur.com/J0T8uCO.jpg)

olikraus, 2017-02-03:
> Vielen Danken Oliver!, spitzes Support!, Das Problem is gelöscht.
:-)

> However I put the former values back and, interesting enough, the pixels are still off...

You need to turn off and disconnect the Arduino for 1 min or so between the test. I assume that the cleard pixel are there from the previous sketsch. 
I mean, the 2/2 will clear the leftmost pixel column. When going back to 1/3 you will see the cleared picture from the previous sketch. The correct setting should be 2/2.

olikraus, 2017-02-03:
I have added the display to u8g2.
You can download the latest version from here:
https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Use the manuall library install method from the Arduino IDE.

//U8X8_ST7565_ERC12864_4W_SW_SPI u8x8(/* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
//U8X8_ST7565_ERC12864_4W_HW_SPI u8x8(/* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);

//U8G2_ST7565_ERC12864_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
//U8G2_ST7565_ERC12864_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);

I would be especially interessed whether the u8x8/FlipMode example works: The inverted text boxes should start exactly at the left/right edge.

joaqrus, 2017-02-04:
Hi Oliver, I am sorry, but with the FlipMode sketch and making many tests, I have realized that the correct settings for this display are:

`` /* default_x_offset = */ 0, 
`` /* flipmode_x_offset = */ 4, 

With this configuration the images (normal and flip) are correctly alligned, as you can see:
[Imgur](http://i.imgur.com/dcz0vfR.jpg)
[Imgur](http://i.imgur.com/GOMX0B9.jpg)
[Imgur](http://i.imgur.com/620csyT.jpg)

The text boxes stay exactly as expected!

olikraus, 2017-02-04:
Thanks for the feedback. Great work! 

I have updated the master lib here:
https://github.com/olikraus/U8g2_Arduino/archive/master.zip

closed 2017-02-08 by olikraus