olikraus/u8g2 #178: setContrast no effect

Arduinolibrary, 2017-02-04:
Hi there,

I met a problem with u8g2, the screen shows characters well, but the its background color are almost same to the words one. it can't be read from in a positive way. Only when you turn around the screen, and watch it from edgewise, you can see the characters.

I adjust the contrast, but no effect, 0 is better, but still too dim. Everything works well on u8glib, but I need to use it on my Genuino 101. So that will be a problem.
U8G2_ST7565_ERC12864_F_4W_SW_SPI
Arduino IDE 1.6.10

olikraus, 2017-02-04:
Can you send a picture?

olikraus, 2017-02-04:
One more remoark: The range is 0 to 255 for setContrast()

olikraus, 2017-02-08:
any feedback?

Arduinolibrary, 2017-02-10:
Sorry, It is U8GLIB_NHD_C12864 u8g(13, 11, 10, 9, 8);
This is U8glib with constrast 0
[image]

This is U8g2 with "u8g2.setContrast(0)"
[image]
[image]
 Actually, I want to use the lcd with Arduino 101, so I have to use u8g2. Plz help

olikraus, 2017-02-10:
Please download the latest u8g2 version from here:

https://github.com/olikraus/U8g2_Arduino/archive/master.zip

Install the via Arduino IDE (Add Zip Library)

In the example you will find the new constructor for NHD_C12864. Please try this new constructor.

olikraus, 2017-02-10:
please also read issue #186

Arduinolibrary, 2017-02-13:
Thank you for your update. it works well on UNO and Mega now.  Like what Cray said, it is blank first. But after I set the contrast to 200. it works!

But now, I met a new problem. It doesn't work with Arduino 101. In the last version, it is just dim, but the new version shows blank. No matter how I set the contrast.

olikraus, 2017-02-13:
Does it still work on Uno and Mega Boards? Maybe this is a shield problem: I mean, maybe the shield is not compatible with the 101.

Arduinolibrary, 2017-02-13:
Suddenly, it works!!!!!!!!! But I don't know why......

Arduinolibrary, 2017-02-13:
This is a good news! Tested it with DFRobot LCD12864 shield.

Arduinolibrary, 2017-02-13:
Thank you for everything! It is amazing!

closed 2017-02-13 by Arduinolibrary