olikraus/u8g2 #180: avoid flashing during wake up (SSD1306 64*48 OLED)

olikraus, 2017-02-05:
> Ah, I must've misread when I commented, mine is an SSD1306 64*48 OLED.
> Is it possible to avoid flashing those on and off when coming back from deep sleep?

ric2b, 2017-02-06:
By the way, I already checked that it's not a power issue. 
If I press the reset button the screen does not flash off and on until the code finally gets to the `u8x8.initDisplay();` line, a few seconds later.

I also noticed that I'm using V1 of the library, do you think there'd be any difference in this behavior by using V2 or is the init sequence functionally identical on both?

olikraus, 2017-02-06:
Oh, you mean the flashing during initDisplay()?
Yes, that was a fundamental change and one of the reasons for writing V2. 
Actually the flashing during start up was completly removed in U8g2 compared to U8glib.

Please crosscheck with u8g2.

ric2b, 2017-02-06:
Ok, I'm trying u8g2 but it still flashes off and on.
This is the shortest code I can make to demonstrate the issue happening, is there something I'm doing wrong?

Thank you

```Arduino
#include <Arduino.h>
#include "U8g2lib.h"

U8G2_SSD1306_64X48_ER_F_HW_I2C u8g2(/*rotation =*/ U8G2_R0);

const int sleepTimeS = 5;

void setup(void)
{ 
  delay(1000);
  
  //u8g2.begin();
  u8g2.initDisplay();
  u8g2.setPowerSave(0);

  u8g2.firstPage();
  do {
    u8g2.setFont(u8g2_font_helvB08_tf);    
    u8g2.drawStr(0,8,"HELLO");
  } while ( u8g2.nextPage() );
  
  Serial.println("ESP8266 going to sleep mode");
  // Must connect D0 to RST. When it wakes up, the device resets and memory is lost.
  ESP.deepSleep(sleepTimeS * 1000000);
}

void loop(void)
{
  
}
```

olikraus, 2017-02-06:
hmmm... will it flash with this code also?

```Arduino
#include <Arduino.h>
#include "U8g2lib.h"

U8G2_SSD1306_64X48_ER_F_HW_I2C u8g2(/*rotation =*/ U8G2_R0);

const int sleepTimeS = 5;

void setup(void)
{ 
  delay(1000); // To demonstrate that the problem isn't caused by power loss from the reset

  u8g2.begin();
  //u8g2.initDisplay();
  //u8g2.setPowerSave(0);

  u8g2.firstPage();
```

the difference is, that the display RAM is cleared after the init phase.

I mean the correct init code is this:

```Arduino
#include <Arduino.h>
#include "U8g2lib.h"

U8G2_SSD1306_64X48_ER_F_HW_I2C u8g2(/*rotation =*/ U8G2_R0);

const int sleepTimeS = 5;

void setup(void)
{ 
  delay(1000); // To demonstrate that the problem isn't caused by power loss from the reset

  //u8g2.begin();
  u8g2.initDisplay();   // send init sequence to display, but keep display in sleep state
  u8g2.clearDisplay();  // clear RAM in the display
  u8g2.setPowerSave(0);  // wake up display, all pixel are off now

  u8g2.firstPage();
```

ric2b, 2017-02-07:
It does, yes :/

olikraus, 2017-02-08:
Hmm... i can not reproduce the problem. Of course the display turns off for a moment. But this is very much required, because the initDisplay() will sent a reset to the display.

If you know, that the display is still active, then you could just skip initDisplay and clearDisplay:

```
void setup(void)
{ 
  delay(1000); // To demonstrate that the problem isn't caused by power loss from the reset

  //u8g2.begin();
  if ( isColdStart )
  {
    u8g2.initDisplay();   // send init sequence to display, but keep display in sleep state
    u8g2.setPowerSave(0);  // wake up display, all pixel are off now
  }
  u8g2.firstPage();
```

`isColdStart` should be 1 if the reset was caused by a power loss, but could be 0 if the reset was just caused by the reset button.

U8g2 can not do anything here, because u8g2 does not know the state of the display after reset and has to do a proper reinit of the display (which causes the flicker). 
Your program has to store the state of the display.

ric2b, 2017-02-08:
Ah, that's it! :)

So my best idea for how to feed isColdStart was to make it read a pin that I can short when I want to restart the screen, otherwise it assumes it's waking from deep sleep.
Don't know if you have a better idea that doesn't involve manually doing this but from what I've seen there's no way for the eps8266 to tell you if it's doing a cold start or waking from deep sleep, since it's just a reset.

Anyway, that's not related to your great library, thanks for the help and patience! :)

olikraus, 2017-02-15:
ok, guess we can close this... I think it is more like a uC specific problem instead of a u8g2 lib problem.

closed 2017-02-15 by olikraus