olikraus/u8g2 #183: 1/4 of sreen works

fire1, 2017-02-07:
Is this a problem in my display or is it in library ? This is from "GraphicTest" ...
[image]
[image]
[image]
[image]

Display's datasheet
[http://www.buydisplay.com/download/manual/ERM24064-1_Series_Datasheet.pdf](url)

olikraus, 2017-02-07:
Which constructor did you use? Did you modify the example?

fire1, 2017-02-07:
Pictures are made from this: 
```
U8G2_T6963_240X64_2_8080
        u8g2(U8G2_R0, 8, 9, 10, 11, 4, 5, 6, 7,
        /*WR*/ 14, /*CE*/ 16, /*dc8*/17, /*RST*/ 18);
```

But this constructor   ....
 ```
U8G2_T6963_240X64_1_8080
        u8g2(U8G2_R0, 8, 9, 10, 11, 4, 5, 6, 7,
        /*WR*/ 14, /*CE*/ 16, /*dc8*/17, /*RST*/ 18);
```
its make even smaller seeing part

olikraus, 2017-02-07:
Can you upload the sketch?

fire1, 2017-02-07:
`
/*

  GraphicsTest.ino

  Universal 8bit Graphics Library (https://github.com/olikraus/u8g2/)

  Copyright (c) 2016, olikraus@gmail.com
  All rights reserved.

  Redistribution and use in source and binary forms, with or without modification, 
  are permitted provided that the following conditions are met:

  * Redistributions of source code must retain the above copyright notice, this list 
    of conditions and the following disclaimer.
    
  * Redistributions in binary form must reproduce the above copyright notice, this 
    list of conditions and the following disclaimer in the documentation and/or other 
    materials provided with the distribution.

  THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND 
  CONTRIBUTORS "AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, 
  INCLUDING, BUT NOT LIMITED TO, THE IMPLIED WARRANTIES OF 
  MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE 
  DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR 
  CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, 
  SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT 
  NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; 
  LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER 
  CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, 
  STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) 
  ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF 
  ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.  

*/

#include <Arduino.h>
#include <U8g2lib.h>

//#ifdef U8X8_HAVE_HW_SPI
//#include <SPI.h>
//#endif
//#ifdef U8X8_HAVE_HW_I2C
//#include <Wire.h>
//#endif

/*
  mega 2560 
  ________________________________________________________________
   T6963c LCD PIN   |    LCD PIN NAME  |     MEGA 2560            |
  ________________________________________________________________|
          1         |          FG      |       GND                |
          2         |         GND      |       GND                |
          3         |         VDD      |       +5V                |
          4         |          VO      |  10K POT WIPER – LEG 2   |
          5         |          WR      |        17                |
          6         |          RD      |        18                |
          7         |       CE/CS      |        14                |
          8         |         C/D      |        15                |
          9         |         RST      |        16                |
         10         |         DB0      |         8                |
         11         |         DB1      |         9                |
         12         |         DB2      |        10                |
         13         |         DB3      |        11                |
         14         |         DB4      |         4                |
         15         |         DB5      |         5                |
         16         |         DB6      |         6                |
         17         |         DB7      |         7                |
         18         |          FS      |       GND                |
         19         |         VEE      |    10K POT–LEG1          |
[... 229 lines cut ...]
    u8g2.begin();
    u8g2.setFont(u8g2_font_6x10_tf);
    u8g2.setFontRefHeightExtendedText();
    u8g2.setDrawColor(1);
    u8g2.setFontPosTop();
    u8g2.setFontDirection(0);
    delay(1000);
}

void loop(void) {
    // picture loop
    u8g2.clearBuffer();
    draw();
    u8g2.sendBuffer();

    // increase the state
    draw_state++;
    if (draw_state >= 9 * 8)
        draw_state = 0;

    // deley between each page
    delay(100);

}
`

olikraus, 2017-02-07:
Just using my mobile. So a quick answer only: you must use F constructor with clear/send buffer.
With 1/2 constructor use the first next loop

Please check with one of the U8g2 examples. These examples should  work.

fire1, 2017-02-07:
OK and thanks! Tomorrow I will reply the result

fire1, 2017-02-08:
I tested like you said with First / Next loop

`  u8g2.firstPage();
  do {
    draw();
    } while ( u8g2.nextPage() );
`
 most of examples from master repo 
,but strangely only GraphicTest is only displaying (without any problems)

olikraus, 2017-02-15:
Not sure what todo next here...

closed 2017-02-17 by fire1