olikraus/u8g2 #185: Font fixing

gborba, 2017-02-08:
Hi. How can I edit fonts? Some numbers are misaligned vertically.
Thanks.

olikraus, 2017-02-08:
The font source has to be fixed. However fixing TTF files will not be possible so easy. Which font glyph has to be fixed?

gborba, 2017-02-08:
I'll take a look. But cant I edit fonts in source code? Just get some
pixels up ir down. I Just cant get to understand the code to bitmap
translation. Sorry, nas english.
Thanks.

olikraus, 2017-02-09:
Can you tell me an example? Can you give me the font name?

gborba, 2017-02-09:
Yep. The u82g_font_fub53_tn for example.
I noted that the glyphs 1, 4, 5, 7, and : have the baseline 1 pixel lower
than others.
I'm using a display with large pixels, so it is noticeable.

Can I  attach images here?

Thanks.

*Gustavo*

olikraus, 2017-02-09:
I am confused, there is no fub53 in u8g2.
See here: https://github.com/olikraus/u8g2/wiki/fntgrpfreeuniversal

ok,let me say this: U8g2 uses two different sources for the font:
 - Special pixel/bitmap fonts
 - True Type Fonts

bitmap fonts will have a good quality, but are avilable only up to 25 pixel height.
TTF fonts are available for any height but are not opimizted for black/white graphics. In general TTF behave unpredictable and such 1 pixel offset is very normal. Unless the font author has specified something else, this is caused by the raster process: The raster tool is otf2bdf, which is another opensource tool. So the offset is caused by otf2bdf. Actually i can not do anything here. It might be thinkable to modify the TTF font itself, but this is beyond my skills.

The font is available here:
https://github.com/olikraus/u8g2/blob/master/tools/font/ttf/FreeUniversal-Bold.ttf

gborba, 2017-02-09:
> ttf/FreeUniversal-Bold.ttf

Sorry, my mistake. It´s fub35 the font name.

Right, but in source code, aren't all fonts stored as bitmap?

Thanks.

*Gustavo*

olikraus, 2017-02-09:
In a TTF file, the glyphs are represented by their outline:

[image]

otf2bdf converts this to a graylevel or black/white bitmap. As you can imagine, a black/white representation may always include rounding errors.

Maybe better user Adobe-X11 fonts:These fonts are manually optimized bitmap fonts.

gborba, 2017-02-09:
Understand that. But cant I edit the resulting bitmap manually in the
source code?.

olikraus, 2017-02-10:
1. Generate the bdf file with this command:
```otf2bdf -a -r 72 -p 49 FreeUniversal-Bold.ttf > fub35.bdf```

2. Edit fub35.bdf (bdf is a well described plain text format --> pls google for it)

3. Use bdfconv to generate the u8g2 format:
```bdfconv -v -f 1 -m '32,48-128' fub35.bdf -o u8g2_my_fub35.c  -n u8g2_my_fub35```

Both tools are located here:
https://github.com/olikraus/u8g2/tree/master/tools/font

gborba, 2017-02-10:
Thanks. I´ll try.

*Gustavo*

olikraus, 2017-02-15:
please reopen for further questions

closed 2017-02-15 by olikraus