olikraus/u8g2 #186: U8g2lib port for ST7565 NHD C 12864

CrayCrayy, 2017-02-09:
Hi olikraus,

Firstly, thank you for all your hard work with this library - it's fantastic.

I am also trying to use an st7565_nhd_c12864, and the st7565_nhd_c12832 constructor works perfectly except for the code not knowing that there are actually 64 pixels instead of 32 in the vertical column. I can even write to the 4th to 7th "tiles" on the lower half of the screen. However, u8x8.begin() does not clear the lower half of the screen.

I was not sure if I should start a new issue, since dingdinglhz wanted something similar but the issue was already closed, so apologies for double posting.

I have tried porting the st7565_nhd_c12864 using the st7565_nhd_c12832  code as a base, following the new display tutorial you wrote. However I have just started learning C++, and playing with Arduinos, and do not know how to do the make build step. I I copied the 12832 constructor your example HelloWorld sketch, and changed the 32 to 64, leaving the other code intact. When I try to compile it, it gives me this error (Arduino IDE 1.8.1).

`error: 'u8x8_d_st7565_nhd_c12864' was not declared in this scope`

which I traced to my addition of

`u8x8_Setup(getU8x8(), u8x8_d_st7565_nhd_c12864, u8x8_cad_001, u8x8_byte_arduino_4wire_sw_spi, u8x8_gpio_and_delay_arduino);` in the U8x8lib.h file - see below for the code I added

I have updated the header and c files by adding the following directly underneath the original:
in U8x8lib.h

`class U8X8_ST7565_NHD_C12864_4W_SW_SPI : public U8X8 {
  public: U8X8_ST7565_NHD_C12864_4W_SW_SPI(uint8_t clock, uint8_t data, uint8_t cs, uint8_t dc, uint8_t reset = U8X8_PIN_NONE) : U8X8() {
    u8x8_Setup(getU8x8(), u8x8_d_st7565_nhd_c12864, u8x8_cad_001, u8x8_byte_arduino_4wire_sw_spi, u8x8_gpio_and_delay_arduino); // **This line is causing the compiler to throw the error**
    u8x8_SetPin_4Wire_SW_SPI(getU8x8(), clock, data, cs, dc, reset);
  }
};`

in u8x8_d_st7565.c - copied the 128632 definitions, renaming the section /* NHD-C12864 */ and changed:
1 -  `/* pixel_height = */` from 32 to 64,
2 - `static const uint8_t u8x8_d_st7565_nhd_c12832_init_seq[] = {` to `static const uint8_t u8x8_d_st7565_nhd_c12864_init_seq[] = {`
3 - `uint8_t u8x8_d_st7565_nhd_c12832` to `uint8_t u8x8_d_st7565_nhd_c12832`
4 - `&u8x8_st7565_128x32_display_info` to `&u8x8_st7565_128x64_display_info` in the first case
5 - `u8x8_d_st7565_nhd_c12832_init_seq` to `u8x8_d_st7565_nhd_c12864_init_seq` in the second case

Do I need to do the make build step to get it working, or did I miss editing a "32" to "64" somewhere? I can't find any other code that has c12832 references in it, and thought that since I already added the definition in the .c file, it should not be telling me I did not declare it?

Thank you

olikraus, 2017-02-09:
Adding a new device is a little bit tricky. I have added a new constructors, which are based on the 128x32 nhd device:
U8G2_ST7565_NHD_C12864_1_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
U8G2_ST7565_NHD_C12864_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);

Of course we have to test and update the code for this constructor (the problem is, that i do not own a C12864 display)

You can download the new u8g2 version here:
https://github.com/olikraus/U8g2_Arduino/archive/master.zip

It can be manually installed into the Arduino IDE via "add lib zip file" menu.

It would be great, if you could have a look at this device. The code for this device is located in u8x8_d_st7565.c, startin at line 404. I just took over the code from the c12832 device (like you suggested). But there might be some changes required in the init sequence or the info struct.

CrayCrayy, 2017-02-09:
Thank you for the super quick response! I've tested the new version, and it works exactly as expected now =D

olikraus, 2017-02-09:
Can you also test the u8x8/FlipMode example?
In both orientations the inverted text must be exactly next to the display boundary.
This would be a great help. Thanks in advance!

CrayCrayy, 2017-02-09:
Yes, actually I already did! I forgot to mention that I needed to include
`u8x8.setFlipMode(1);`
and
`u8x8.setContrast(150);`
in setup( ) { }, otherwise the screen appears blank, and is upside down when first initialized.

I just tested the FlipMode example, and it works exactly as it should.

Is there anything else you would like me to test on it? I'll be more than happy to.

In case this information is useful to you, the unit I'm running is on the DFRobot DFR02887 lcd shield V1.0 with 5 way joystick from here:
[https://www.dfrobot.com/index.php?route=product/product&product_id=1084](url).
The stock library that is recommended is your original U8glib - I couldn't find anything links at all for using the U8g2lib using Google, and had to try a few different keywords to find @dingdinglhz's post. Perhaps now the distributors can update the stock recommended library =D.

olikraus, 2017-02-09:
Thanks a lot for testing the FlipMode example. Good job.

Regarding the contrast topic. You are saying, that you must use this contrast setting, right?
So you are saying, that you **must** apply the "u8x8.setContrast(150);" command, right?
Wouldn't it be better if the value is included into the setup?

Thanks,
Oliver

CrayCrayy, 2017-02-10:
Yes, I *must* apply the setContrast command after initializing via begin. I think the value the contrast is initialized at is around "100" or below, because the display appears blank at those levels. I wasn't sure if "150" was the best value so I used this short loop to check:

`  int i;
  for (i=255;i>0;i--){
    u8x8.setContrast(i);
    u8x8.drawString(0,1,"                ");
    char buffer[] = "Contrast at    ";
    sprintf(buffer, "Contrast at %d", i);
    u8x8.drawString(0,1,buffer);
    delay(500);
  }`

The display was easily readable at contrast levels between 115-215, and best between 130-190. I'm not sure if the backlight will affect it but it will take quite a while to run those contrasts at all the different backlight levels and tabulate the results. Also, I didn't see a backlight level method on the reference page?

Yes, I agree that it would be better with a default contrast value of 180 included in the setup. The words on the display were most easily seen from all angles with that value, almost 180 degrees.

Would it be difficult to implement a clearLine method? This would be easier than clearing the one line manually by writing " " to all the tiles. I got these ideas from modifying the example sketches and cpp files of the other LCD library I use.

olikraus, 2017-02-13:
ToDo
 * [ ] Put u8x8.setContrast(150) into init code
 * [ ] Think about a clear line method for u8x8

olikraus, 2017-03-07:
added
void u8x8_ClearLine(u8x8_t *u8x8, uint8_t line);

olikraus, 2017-03-07:
documentation updated

closed 2017-03-07 by olikraus

CrayCrayy, 2017-03-08:
Thanks Oliver!

Seann