olikraus/u8g2 #187: Is there native support for the ESP32 ESP-IDF?

nkolban, 2017-02-11:
I'm a big fan of the ESP32 and am about to start studying u8g2 from an ESP32 perspective.  However, I don't want to use the Arduino interfaces but instead want to use the ESP32 ESP-IDF native functions for I2C and SPI.  Is there an existent HAL for the ESP32 ESP-IDF?  If yes, can I get a pointer please.   If no, I'd like to have a go at providing one.  If that would be of interest for this project, please let me know.  If needed, I can be reached by email at kolban1@kolban.com.

olikraus, 2017-02-11:
There is no HAL for the ESP, but there is some documentation how to do this. 

We have started the documentation here:
https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform

Previous discussion happend here: #179 and #117 

Let me know if something is missing or if you have further questions.

nkolban, 2017-02-12:
Thank you sir ... that was pretty painless.  I wrote some handlers for the `msg_gpio_and_delay_cb` and `msg_comms_cb` functions which interpret and process the messages ... I used hardware SPI and hardware GPIO to drive an SSD1306 and it worked just great.  I'll write up in detail.    Do you keep a repository or some other location where others who come to U8G2 could find native ESP32 samples?

olikraus, 2017-02-12:
I am glad that the description was complete enough for you to port U8g2 to your target system.

There is no plan at the moment to list ports to other systems. But yes, it sounds like a good idea.
I guess it would be great for others to take over your callback functions. Maybe you can just document this here in this issue. Additionally I will add links to the issue from the porting guide.

olikraus, 2017-02-12:
I have added the some links here:
https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform

nkolban, 2017-02-12:
I have uploaded the ESP32 HAL to here:

https://github.com/nkolban/esp32-snippets/tree/master/hardware/displays/U8G2

A `README.md` is also present and I am happy to take bug fixes and enhancement requests.

olikraus, 2017-02-13:
Wow, thanks a lot. Amazing video.

olikraus, 2017-03-04:
Guess, this can be closed

closed 2017-03-04 by olikraus

nkolban, 2017-03-04:
If there is anything of the work I did that could be absorbed into the proper u8g2 library ... that would be great.  Ideally your Github repository should be the source of all knowledge and truth relating to U8G2.

olikraus, 2017-03-04:
Do you have anything specific in mind?

nkolban, 2017-03-04:
3 weeks is an eternity in my tinkering ... let me come back to it again and look over what ever I did to get it working.  That way I'll be able to give you an "impact analysis" and "integration suggestions" which would be what I would suggest.  It all hinges on desire and intent.   From what I remember, the code base assumes Arduino APIs.  The ESP32 has its own ESP-IDF APIs.  The tinkering I did created a mapping for the small subset that were needed.  The liklihood would be the creation of a directory to host ESP32 ESP-IDF specific source files and a change (or instructions) to the build system to accommodate this.   As such, and as the owner of this library, the question falls to you as to whether or not you want board specific code in the repository.   If the answer is "no" ... then thats the end of the discussion.   If the answer is "yes", then it would be up to thee and me on how best to achieve that.  You own the repository structure and direction while I would act as your short term consultant to assist you with integration of ESP32 specific work until and unless you yourself came up to speed on that area.

olikraus, 2017-03-04:
OK, I only have mobile to type here at the moment. Please excuse typing errors.

Indeed I do not want uC specific code in the folder U8g2/src .
Then, U8g2 is still a Arduino lib. All Arduino specific code goes to cppsrc. This Arduino specific code might include ESP32 code. In fact I have a ESP 32 board in my lab and I just wait for the ESP32 Arduino port Being stable enough. 

Then there is U8g2/sys folder, which could contain any uC code. For example see sys/arm/lpc824 .
I suggest to create sys/esp32.

Would this make sense?

How ever finally it also depends on you whether you want to share something. Project license is two clause BSD...

nkolban, 2017-03-04:
Make sense.  As for licensing/sharing ... my hope is that any code I produce for inclusion in your repository becomes 100% yours.  I have no need for credit or ownership ... I don't even need nor desire my name in any comments in the code.  Let me look at what we might put in the U8g2/sys folder from an ESP32 perspective plus perhaps a `README.md`.  Then maybe I can for your project, add those items and issue a pull request.  Alternative I can just ZIP them up and you can add the under your own Github id.  Either way is fine for me.

Once included, then I'll delete the ESP32 U8g2 samples from my own ESP32 samples Github repository and there will be only a single source of truth, your U8g2 library.

olikraus, 2017-03-04:
Sounds good. Regarding the license, I just wanted to mention that U8g2 code is also available for commercial use. You should still keep ownership.