olikraus/u8g2 #188: drawXBM() - Hex must be mirrored?

Tysonpower, 2017-02-11:
I am converting a u8g arduino sketch for the new u8g2lib at the moment.

Everything works fine, except the drawBitmap replacement drawXBM.
drawXBM draws every bitmap mirroed, in comparision to the older u8g lib.
Is that the new format or a bug?

I used the following hex values before: {0x18,0x00,0x00,0x00,0x80,0x80,0x81,0xf3}

[image]
To get the right bitmap on the oled: {0x18,0x00,0x00,0x00,0x01,0x01,0x81,0xcf}
[image]
 
Code (Helloworld example edited):
```
#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

U8G2_SSD1306_128X64_NONAME_1_SW_I2C u8g2(U8G2_R0, /* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE);   // All Boards without Reset of the Display

// End of constructor list

static unsigned char bitmapCheck[] = {0x18,0x00,0x00,0x00,0x01,0x01,0x81,0xcf};

void setup(void) {
  u8g2.begin();  
}

void loop(void) {
  u8g2.firstPage();
  do {
     u8g2.drawXBM(10, 40, 8, 8, bitmapCheck);
  } while ( u8g2.nextPage() );
}
```

olikraus, 2017-02-12:
There is a function
```
 void drawBitmap(u8g2_uint_t x, u8g2_uint_t y, u8g2_uint_t cnt, u8g2_uint_t h, const uint8_t *bitmap)
```
also in U8g2. This should be compatible with U8glib. As you encountered, drawXBM is **not** compatible.

Tysonpower, 2017-02-12:
Thanks for the reply.

DrawBitmap is compatible, but i would like to use the new XBm function. You even wrote a note in the wiki: 

> This function should not be used any more, please use drawXBM instead.

Is there a bitmap creator for XBm out there?`before i was able to use [dot2pic](http://dot2pic.com/) to create bitmaps.

regadrs,
Manuel

olikraus, 2017-02-12:
Yes, you can use any tool which exports xbm, for example GIMP.

olikraus, 2017-02-15:
closing this... pleae reopen for any further questions...

closed 2017-02-15 by olikraus