olikraus/u8g2 #189: alternating lines blank

Ariffffff, 2017-02-12:
I have bought this cheap oled 0.96 inch IIC from aliexpress, its showing alternating lines blank .

Is there a way to check if module is faulty or some coding problem. (so if module is faulty, i dont buy more from same vendor)

I am using this option. 

[image]
U8X8_SSD1306_128X64_NONAME_SW_I2C u8x8(/* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE);   // OLEDs without Reset of the Display

 the image came a bit blurry as camera not focusing light, but you can feel the vertical skipped lines, this happened.
[image]

[image]

Ariffffff, 2017-02-12:
this link might be of help as its saying smothing same, but I have no clue if this is for arduino or what

https://forums.electricimp.com/discussion/2786/ssd1306-oled-128-x-64-display

olikraus, 2017-02-13:
Thanks for puting the issue here.
Please locate file u8x8_d_ssd1306_128x64_noname.c on your local Arduino library folder.
(reference: https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ssd1306_128x64_noname.c)

In line 61 you find this command:
 U8X8_CA(0x0da, 0x012), 

Please change this value to 
 U8X8_CA(0x0da, 0x002), 

Will this solve the problem?
You can also try
 U8X8_CA(0x0da, 0x022), 

and maybe also:
 U8X8_CA(0x0da, 0x032), 

Will any of these values solve the problem?
Thanks

Ariffffff, 2017-02-13:
here are my results.

U8X8_CA(0x0da, 0x012),
[image]

Please change this value to
U8X8_CA(0x0da, 0x002),
visibility gets better, 
[image]

U8X8_CA(0x0da, 0x022),
[image]

U8X8_CA(0x0da, 0x032),
gets more bad

Ariffffff, 2017-02-13:
I think this might be a manufacturing fault. In my next cart from aliexpress I will retry from other vendor.

olikraus, 2017-02-13:
Yes, agree, looks like a hardware problem.

Ariffffff, 2017-02-13:
Thankyou for time and effort

closed 2017-02-13 by Ariffffff