olikraus/u8g2 #190: T6963 240X64 parallel not working with Arduino Mega

ve3jrw, 2017-02-12:
The setup I have T6963 based display and Arduino Mega works fine with the u8g library, but doesn't seem to function with the u8g2 library.

The display just shows garbled pixels with u8g2 while running the Hello World example sketch.

u8g setup is:

U8GLIB_T6963_240X64 u8g(LCD_DB0, LCD_DB1, LCD_DB2, LCD_DB3, 
                        LCD_DB4, LCD_DB5, LCD_DB6, LCD_DB7, 
                        LCD_CE, LCD_CD, LCD_WR, LCD_RD, LCD_RST);

u8g2 setup is:

U8G2_T6963_240X64_1_8080 u8g2(U8G2_R0, LCD_DB0, LCD_DB1, LCD_DB2, LCD_DB3, 
                                  LCD_DB4, LCD_DB5, LCD_DB6, LCD_DB7, 
                                  LCD_WR, LCD_CE, LCD_CD, LCD_RST);

And then I'm using the corresponding example code.  I'd be happy to help debug if you can give me some pointers.

Specifically, it seems odd that you don't have the read line in the u8g2 constructor.  I don't understand how the read can work without this line.

olikraus, 2017-02-13:
Did you connect the RD line to 5V?

ve3jrw, 2017-02-14:
Yes, 

How do you check the display status without using the read signal?  

The documentation indicates the RD line must be low to read the status.  I don't see how this can be done without knowing what the RD line is.

ve3jrw, 2017-02-14:
Found an issue in the initialization that was causing the problem.  Thanks for the response as it helped lead to the solution.

closed 2017-02-14 by ve3jrw

olikraus, 2017-02-14:
OK....