olikraus/u8g2 #191: ST7920 messy lines and points

techscout1, 2017-02-13:
I have an issue with my ST7920 Display. The sketches itself works fine, but sporadically unwanted points and lines appear on the display. With the old u8glib everything works fine, but after the update to u8g2 it doenst work well.

Does somebody have an idea?

My Board: Arduino Mega 2560

I used the constuctor:

//U8G2_ST7920_128X64_F_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* CS=*/ 10, /* reset=*/ U8X8_PIN_NONE);

olikraus, 2017-02-13:
Is your display connected to the HW SPI lines? Can you test the following constructor?

U8G2_ST7920_128X64_F_HW_SPI u8g2(U8G2_R0, / CS=/ 10, / reset=*/ U8X8_PIN_NONE);

techscout1, 2017-02-13:
It needed a litte bit of work to change the layout of the circuit board but it works fine! Thank you for that fast help. u8g2 is one of the greatest libraries 🥇 !! :)

closed 2017-02-13 by techscout1