olikraus/u8g2 #192: Support for mutiple SSD1306/SH1106 displays on I2C

Jeffs4653, 2017-02-13:
I have a project where I would like to support multiple OLED SSD1306 (or SH1106) displays on I2C. There were for hacks to U8GLIB that could be done for this since it didn't support multiple instances (although I never tried them), but I have had to move to U8G2 because I am using an ARM Cortex M0 (Adafruit Feather M0 + WiFi) and the older U8GLIB doesn't work at all on that board.

The SSD1306/SH1106 controller supports two I2C addresses (0x3C/0x3D [or 0x78/0x7A as marked on the generic Chinese displays]), with 0x3C/0x78 being the default.

Is it possible now to define two instances of U8G2 to use two displays, or is this something that can be implemented or are interested in implementing?

Thanks

olikraus, 2017-02-13:
Glad you want to switch to U8g2.
The I2C address topic is now solved in U8g2 and hacks are not required any more:

https://github.com/olikraus/u8g2/wiki/u8g2reference#seti2caddress

Jeffs4653, 2017-02-13:
I tried it with two displays using two objects with different addresses and it works as you said it would. That is great! 

I notice however that the refresh speed is a slow. Much slower than the older u8glib. With the previous library I was using the "NO ACK" definition since these generic OLED displays don't ACK the I2C communication. Could this possibly be why the displays on U82G are refreshing slower?

Is there a NO ACK definition ? I am using:

U8G2_SSD1306_128X64_NONAME_F_SW_I2C screen1(U8G2_R0, SCL, SDA, U8X8_PIN_NONE); 

For reference, I previously used:

U8GLIB_SSD1306_128X64   Screen(U8G_I2C_OPT_NO_ACK);       // Display which does not send ACK

olikraus, 2017-02-13:
You compare the SW emulated I2C from u8g2 with the Hardware implementation from U8glib. In fact the software I2C emulation is very slow. But this is the price for the support of all boards. Anyhow, better use the hardware I2C version of U8g2 for maximum speed:

U8G2_SSD1306_128X64_NONAME_F_HW_I2C screen1(U8G2_R0, U8X8_PIN_NONE);

Jeffs4653, 2017-02-13:
Thank you for correcting an oversight. I wasn't aware the "SW" referred to software emulation. I tried the HW and yes, much faster (speed as am used to). 

Still getting used to the new library. Lots of conversion to do as I move the ARM from the AVR.

olikraus, 2017-02-13:
Maybe documentation is not that clear about the difference between HW and SW in the constructor name. 

I should add this to the FAQ...
https://github.com/olikraus/u8g2/blob/master/doc/faq.txt

Jeffs4653, 2017-02-14:
Having gotten the program to work with two SSD1306 128x64 0.96" displays, I'm trying to move forward with the actual second display I wish to use. This is an Adafruit Featherwing OLED display which is listed as a SSD1306 128x32 I2C. I find no direct constructor for a display of this dimension on the 1306 (I saw one for the SSD1305, but it didn't work properly.)

My desire is to use a primary display of 128x64:
U8G2_SSD1306_128X64_NONAME_F_HW_I2C screen1

and the secondary display of 128x32, which would be something like:
U8G2_SSD1306_128X32_NONAME_F_HW_I2C screen2

As of now the second constructor 'U8G2_SSD1306_128X32_NONAME_F_HW_I2C' does not name a type.

Also, I belive I found some errors in the documentation for the constructors. There are many listed as:
U8G2_SSD1306_128X64_NONAME_F_HW_I2C(**rotation, [reset [, clock, data]]**), but in the actual code I'm using it's "**rotation, clock, data, reset**"

[https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#ssd1306-128x64_noname](url)

olikraus, 2017-02-14:
The Featherwing OLED is actually a Univision OLED, the constructors are something like this. In the example, the FeatherWing OLED is also named.

//U8G2_SSD1306_128X32_UNIVISION_1_SW_I2C u8g2(U8G2_R0, /* clock=*/ 21, /* data=*/ 20, /* reset=*/ U8X8_PIN_NONE);   // Adafruit Feather M0 Basic Proto + FeatherWing OLED
//U8G2_SSD1306_128X32_UNIVISION_1_SW_I2C u8g2(U8G2_R0, /* clock=*/ SCL, /* data=*/ SDA, /* reset=*/ U8X8_PIN_NONE);   // Adafruit Feather ESP8266/32u4 Boards + FeatherWing OLED
//U8G2_SSD1306_128X32_UNIVISION_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);   // Adafruit ESP8266/32u4/ARM Boards + FeatherWing OLED
//U8G2_SSD1306_128X32_UNIVISION_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE, /* clock=*/ SCL, /* data=*/ SDA);   // pin remapping with ESP8266 HW I2C

> U8G2_SSD1306_128X64_NONAME_F_HW_I2C(rotation, [reset [, clock, data]]), but in the actual code I'm using it's "rotation, clock, data, reset"

Yes, this was newly introduced for the ESP8266 --> #158

Jeffs4653, 2017-02-14:
Using:
U8G2_SSD1306_128X32_UNIVISION_1_HW_I2C screen1(U8G2_R0, SCL, SDA, U8X8_PIN_NONE);

The display is addressed and data appears on it, but the coordinate system seems messed up. 0,0 doesn't appear in the upper left corner. All data appears to be displaying onn the bottom edge. I tried all the rotation angles but that didn't correct it (it rotated but was still off the display).

This code, when used with the 128x64 display correctly showed a line of text top left
```
screen2.setFont(u8g2_font_6x13_tf);   
screen2.drawStr(0,10, buf);
screen2.sendBuffer();
```
With the 128x64 I see this:

[image]

Jeffs4653, 2017-02-14:
Using:
U8G2_SSD1306_128X32_UNIVISION_1_HW_I2C screen1(U8G2_R0, SCL, SDA, U8X8_PIN_NONE);

The display is addressed and data appears on it, but the coordinate system seemsmessed up. 0,0 doesn't appear in the upper left corner. All data appears to be displaying onn the bottom edge. I tried all the rotation angles but that didn't correct it (it rotated but was still off the display).

This code, when used with the 128x64 display correctly showed lines of text top left
```
    screen1.clearBuffer();                    // clear the internal memory
    screen1.setFont(u8g2_font_6x13_tf);   // choose a suitable font
    screen1.drawStr(0,10,"SSID: ");    // write something to the internal memory
    screen1.drawStr(35,10,WiFi.SSID());
    screen1.drawStr(0,25,"Signal: ");    // write something to the internal memory
    snprintf (buf, 128, "%02d dBm", WiFi.RSSI());
    screen1.drawStr(50, 25, buf);
    screen1.sendBuffer();                    // transfer internal memory to the display

```
With the 128x64 I see this:
[image]

olikraus, 2017-02-14:
I think this is a software issue: You must use the _F_ constructor:
U8G2_SSD1306_128X32_UNIVISION_F_HW_I2C screen1(U8G2_R0, SCL, SDA, U8X8_PIN_NONE);

The _1_ and _2_ constructor requires the firstPage/nextPage loop

Jeffs4653, 2017-02-14:
> The 1 and 2 constructor requires the firstPage/nextPage loop

Ah yes, like I used with U8GLIB.

Anyhow, that fixed it!

I think I'm good to go now.

closed 2017-02-26 by olikraus

walenw, 2017-11-25:
Dear Friends,
Sorry for bothering you with this simple issue, but it iss connected with this thread. I have two OLED SH1106 displays connected via I2C. I have changed physical adress resoldering apropiate pins on the bottom side of displays. Now I have two devices on I2C bus - 0X3C and 0X3D. The problem is that I can not use two displays the same time. What can I do?

[image]

`#include "U8g2lib.h"
#include <Wire.h> //Library for I2C interface 

U8G2_SH1106_128X64_NONAME_F_HW_I2C OLED_1(U8G2_R0, U8X8_PIN_NONE);
U8G2_SH1106_128X64_NONAME_F_HW_I2C OLED_2(U8G2_R0, U8X8_PIN_NONE);

void setup() {
  u8g2_SetI2CAddress(&OLED_1, 0x3C);
  u8g2_SetI2CAddress(&OLED_2, 0x3D);
  OLED_1.begin();
  OLED_2.begin();
  OLED_1.setFont(u8g_font_6x10);
  OLED_2.setFont(u8g_font_6x10);
}

void loop() {
  static unsigned long oledTimer = millis();      //every 1000ms update the oled display
  if (millis() - oledTimer >= 1000)  {
    oledTimer = millis();
    drawOLED_1();
    //drawOLED_2();
  }
}

void drawOLED_1(void) {
  char buffer[10];
  OLED_1.clearBuffer();          // clear the internal memory
  OLED_1.setFont(u8g_font_6x10);
  OLED_1.drawStr(0, 10, "Sensor 1 -> ");
  dtostrf(10, 5, 0, buffer);   //print value 10
  OLED_1.drawStr(80, 10, buffer);
  OLED_1.sendBuffer();         // transfer internal memory to the display
}

void drawOLED_2(void) {
  char buffer[10];
  OLED_2.clearBuffer();          // clear the internal memory
  OLED_2.setFont(u8g_font_6x10);
  OLED_2.drawStr(0, 10, "Sensor 2 -> ");
  dtostrf(20, 5, 0, buffer); //print value 20
  OLED_2.drawStr(80, 10, buffer);
  OLED_2.sendBuffer();         // transfer internal memory to the display
}`