olikraus/u8g2 #194: List of languages supported

MsiSoftwareEngineer, 2017-02-14:
Is there a list of languages that can be written on the font section detailing what languages are supported. Most of the _xf fonts have a the same set of characters. For example the Hungarian hő is not in any of them (or at least that i could find). Like always, thanks for great library!

olikraus, 2017-02-14:
"xf" font include unicode 32 to 255. Usually languages (in the sense of special unicode ranges) are added on request. Or you just create them by yourself with bdfconv from a suitable font source.
If added to u8g2, then these fonts will have the language name as post fix (_hungarian).

Some examples are here:
https://github.com/olikraus/u8g2/wiki/fntgrpunifont

MsiSoftwareEngineer, 2017-02-16:
Excellent, it looks like we'll just use the available fonts. I have noticed even back in u8glib and now also in u8g2, certain fonts, just crash our software, using both constructors:

`U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI u8g2;	//Full framebuffer - fast but buffer size is 2048 bytes
	U8G2_SSD1322_NHD_256X64_1_4W_HW_SPI u8g2; `

Fonts tried: 

`#define AddedUsedFluidDataFont u8g2_font_fur14_tf //Nice size
//u8g2_font_7x14_tf //14 pixel AC: Kinda smallish but very clear 
// u8g2_font_9x15_tf screen went dark again
//u8g_font_fur17                   //too large QK
//u8g_font_courB18                ///15 pixel screen went dark again - QK
//u8g_font_fur14                              //14 pixel option
//u8g_font_fub11               //11 pixel - bold, ray don't like
//u8g_font_helvB10	//heidi didn't like it`

What could cause certains fonts from not being able to be used? We are using an AtMega2560 processor and the latest version of u8g2.

olikraus, 2017-02-17:
I never had any problems with the fonts except for the size (i mean, the font must fit into the flash ROM).
If you have some code which causes your system to crash, i would be happe to see this.

olikraus, 2017-02-26:
no feedback... closing

closed 2017-02-26 by olikraus

MsiSoftwareEngineer, 2017-02-27:
Sorry, yes that's the first thing I thought of, and I changed the constructor to not use the full buffer but the system blew up regardless. Now our display doesn't like bitmaps if the height or width is not divisible by eight. But I don't think that's related. Currently we just avoiding the trouble fonts but if I found the problem I'll post it here. Thanks,

olikraus, 2017-02-27:
Sure, no problem.