olikraus/u8g2 #195: using different fontSize in same drawing?

zenz, 2017-02-14:
I'd give it a try but cannot make it. While I first set it to bigger font size, and drawStr something, and then set it to smaller font size and drawStr something.  the bigger strings will move up to the top of the screen.

If it's possible, can you show a sample how to make it work?

Best regards

olikraus, 2017-02-14:
can you post the related non working code?

zenz, 2017-02-14:
please check below code.

#include "U8g2lib.h"
#include <Wire.h>

U8G2_SSD1306_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ 12, /* data=*/ 14, /* reset=*/ U8X8_PIN_NONE);

void setup() {
  // Init oled
  u8g2.begin();
  u8g2.setDrawColor(1);
  u8g2.setFontPosTop();
  u8g2.setFontDirection(0);
}

void loop() {
    u8g2.clearBuffer();
    u8g2.setFont(u8g2_font_logisoso34_tf);
    u8g2.drawStr( 5, 12, "20");
    u8g2.drawStr( 54, 12, "|");
    u8g2.drawStr( 79, 12, "40");
    // here mass the display
    u8g2.setFont(u8g2_font_ncenB14_tr);
    u8g2.drawStr(10,10,"T");
    u8g2.sendBuffer();
}

olikraus, 2017-02-14:
Can you post a picture also?

olikraus, 2017-02-14:
Just to ensure, that this is not a font mode issue:
Did you play around with this command also?
u8g2_SetFontMode(u8g2_t *u8g2, uint8_t is_transparent);

https://github.com/olikraus/u8g2/wiki/u8g2reference#setfontmode

zenz, 2017-02-14:
[image]
the above makes a correct display with only one font

[image]
this one like a mass if using two different font

zenz, 2017-02-14:
it's not the issue of setFontMode, please check above post for the image.

zenz, 2017-02-14:
I'm using NodeMCU(ESP8266-12E) as MCU

olikraus, 2017-02-14:
I agree, this looks odd.
Let me try to reproduce the problem on my own systems.

olikraus, 2017-02-14:
I think the software behaves correctly and there is no problem. The only issue is, that setFont will reset the setFontPosTop command. This means, you need to call u8g2.setFontPosTop(); after each font change.
```
void loop() {
u8g2.clearBuffer();
u8g2.setFont(u8g2_font_logisoso34_tf);
u8g2.setFontPosTop();
u8g2.drawStr( 5, 12, "20");
u8g2.drawStr( 54, 12, "|");
u8g2.drawStr( 79, 12, "40");
// here mass the display
u8g2.setFont(u8g2_font_ncenB14_tr);
u8g2.setFontPosTop();
u8g2.drawStr(10,10,"T");
u8g2.sendBuffer();
for(;;)
  ;
}
```
I agree, that is should either document this behavior or change the behavior of the code itself.

olikraus, 2017-02-14:
ok, i have changed the code

https://github.com/olikraus/u8g2/commit/51983b5485a4cf8cd599a49badd6b213d4d97a3a

Thanks for this issue.

zenz, 2017-02-15:
thanks for solving this bug so quick!

zenz, 2017-02-15:
Sorry that I had to re-open this issue.
It fixs the problem which cannot use different fontSize in same drawing, but creates another problem.

While using the new solution, the display will be freeze... without any update.

olikraus, 2017-02-15:
Any example code which demonstrates the freezing?

olikraus, 2017-02-15:
Did you update the init code also? 

I have placed the beta release here:
https://github.com/olikraus/U8g2_Arduino/archive/master.zip

Install the zip lib via the Arduino IDE Add Zip Lib menu.

zenz, 2017-02-15:
Sorry, my fault, I forgot to modify the u8g2_setup.c file together, now all things running smooth.
Thank you for fixing the bug again!

closed 2017-02-15 by zenz