olikraus/u8g2 #199: Implementation of uC specific functions

hieuvt6-fti, 2017-02-20:
A questions regarding the two functions that could be ported for a particular microcontrollers.

To my understanding, many of the delays and GPIO push/pull messages that "uC Specific" has to handle is for the software bit-bang interface.

My MCU has its own dedicated I2C peripheral. If I implement my own Communication Callback functions, would that mean there will be certain messages in "uC Specific" that I won't have to implement? If so, what are they?

olikraus, 2017-02-20:
Did you read this wiki page?
We tried to put together all the required information there:

https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform

hieuvt6-fti, 2017-02-21:
Yes, I have.  But as I explained, I would like to avoid using the bit-bang solution, and consequently would like to not have to implement the unnecessary messages that the "uC Specific" functions would otherwise need to handle.

As a request, I think that it would be really neat if this kind of information is included in that wiki page as well. For example, "if you implement your own SPI function, you only need to handle the following messages {...}".

hieuvt6-fti, 2017-02-21:
Another thing is the two messages `U8X8_MSG_DELAY_NANO` and `U8X8_MSG_DELAY_100NANO` in particular. That is basically impossible with my STM32L011K4 running at maximum clock of 32MHz,

To figure out a way to implement the delays, I have the infinite while loop calling only a GPIO toggling function and `__nop()` like this

```
while (1)
{
  HAL_LPTIM_Counter_Start_IT(&hlptim1, 24900);
  __nop();  __nop();  __nop();  __nop();  __nop();
}
```

What I found is that even with no `__nop()` calls I could only toggle the GPIO once every 490ns.

olikraus, 2017-02-21:
You can ignore both nano delay messages. The are not used in U8g2.

For the SPI transfer, pls have a look at the byte transfer functions.

closed 2017-02-26 by olikraus