olikraus/u8g2 #202: Switching SPI to I2C (or vice versa) dynamically

hazarkarabay, 2017-02-23:
Hi,

This is a question, not an "issue". I could not find any other place to ask. 

I need to determine connected display type at boot time and initialize it accordingly. The display (SSD1306) can communicate via SPI or I2C. 

I'm able to detect connection type at startup with a simple logic. If I don't receive the I2C ACK's on specified addresses, I assuming it is connected via SPI.

```
// Determining display type
  Wire.begin();
  Wire.beginTransmission(0x3C);
  byte tmp = Wire.endTransmission(true);

  if (tmp == 0) {
    // I2C on 0x3C
  } else {
    Wire.beginTransmission(0x3D);
    tmp = Wire.endTransmission(true);
    if (tmp == 0) {
      // I2C on 0x3D
    } else {
      // SPI
    }
  }
```

The problem is, currently I need to know the required constructor name before setup(). In other words, I can't place the constructor in setup or anywhere else. 

How can I accomplish this dynamic defining? Thanks.

olikraus, 2017-02-24:
First you define two different devices, u8g2_a and u8g2_b for SPI and I2C.
Depending on your detected type of display, you copy the address from one of the two to a pointer variable. After that you use the pointer variable to access the correct display. Instead of the "." notation you use the "->".

```
U8G2_SSD1306_128X64_NONAME_1_4W_HW_SPI u8g2_a(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
U8G2_SSD1306_128X64_NONAME_1_HW_I2C u8g2_b(U8G2_R0,/* reset=*/ 8);
U8G2 *u8g2;
...
if ( tmp == 0 ) {
  u8g2 = &u8g2_b;
} else {
  u8g2= &u8g2_a;
}
u8g2->begin();
...
```

hazarkarabay, 2017-02-24:
Thank you, it worked well. 

One thing I noticed when I used this two constructor and pointer solution, available memory decreases. Probably inevitable, but still I'll ask if there is a way to doing this without using extra memory (or less memory).

Normal usage (with one constructor)
```
Sketch uses 11056 bytes (34%) of program storage space. Maximum is 32256 bytes.
Global variables use 812 bytes (39%) of dynamic memory, leaving 1236 bytes for local variables. Maximum is 2048 bytes.
```

Dynamic variant (same as your example with hardcoded "u8g2 = &u8g2_b;")
```
Sketch uses 12016 bytes (37%) of program storage space. Maximum is 32256 bytes.
Global variables use 949 bytes (46%) of dynamic memory, leaving 1099 bytes for local variables. Maximum is 2048 bytes.
```

olikraus, 2017-02-26:
Hmm... you could replicate the constructor code. Maybe this works:

```
U8G2 u8g2;
if ( is_i2c ) {
    u8g2_Setup_ssd1305_i2c_128x32_noname_f(u8g2.getU8g2(), rotation, u8x8_byte_arduino_hw_i2c, u8x8_gpio_and_delay_arduino);
    u8x8_SetPin_HW_I2C(u8g2.getU8x8(), reset, clock, data);
} else {
    u8g2_Setup_ssd1306_128x64_noname_1(u8g2.getU8g2, rotation, u8x8_byte_arduino_hw_spi, u8x8_gpio_and_delay_arduino);
    u8x8_SetPin_4Wire_HW_SPI(u8g2.getU8x8(), cs, dc, reset);
}
```

closed 2017-02-27 by hazarkarabay