olikraus/u8g2 #203: A u8g2 library bug or  IDE bug about UTF-8  format Display.

qnxsgwy, 2017-03-01:
I directly open a u8g2 example in IDE and revise it,UTF-8 format can't be  displayed.But after save the example to another addresses, UTF-8 format can be display.

olikraus, 2017-03-01:
What means addresses?

VitaliGN, 2017-03-05:
I think I understand what he meant by "addresses" it is a code of letter. I have got some kind of such problem also. 
Just for example: the font - "u8g2_font_nine_by_five_nbp_tf" (you can see it here: [https://github.com/olikraus/u8g2/wiki/fntgrpnbp#nine_by_five_nbp](url) ) has Cyrillic letters but if I type Cyrillic letter for example 'П' from my keyboard (you can see this letter in the font last on the line =1040/0410  it shows letter of some other language 'Ð' (this letter is the first in line 208/00d0). Right now I am in the middle of finding out why it happens. 
The codes (addresses) of these letters correspond to unicode hex encoding. So I think about my compiler now. probably it has some other encoding which is not unicode and that is why letter 'П' shows on the display as 'Ð'. I was thinking about making a function which would convert strings into right codes of letters, but that is bad. there should be another solution may be I should change the encoding of compiler, trying to find out how to do all this. I also need to make another 2 fonts which is not here. May be same as "u8g2_font_nine_by_five_nbp_tf" but
1st font) one pixel lower 
2nd font) 2 pixels lower and 1 pixel thinner 
By the way, there in the description of "u8g2_font_nine_by_five_nbp_tf" it says that it's width is 10 and height is 15. But in fact capital letter 'V' is 5 pixels wide and 9 pixels high.

**So to sum up my big text of not understanding:**
1) I think there is a conflict in encoding of the font and encoding of compiler
2) I also think that to fit the font "u8g2_font_nine_by_five_nbp_tf" i will have to change the encoding of compiler.
3) Or I could change those letters which are typed in the font to fit to my keyboard (or encoding)
4) I should make another new font for my needs but that I am discussing yet in the related issue #204 .

olikraus, 2017-03-06:
The compiler can be modified via 

-finput-charset=charset

option. This option can be also applied in the Arduino IDE (if i remember correctly). It should be something like "-finput-charset=UTF-8".

gcc also tries to derive the charset from the current "locale". Maybe this has caused the problem, if this is not set to UTF-8.

VitaliGN, 2017-03-11:
I am using Visual Studio and I understand that it is not for this thread, but I have tried to change the source encoding to various encodings including unicode, utf-8 with bom and without, but now the compiler does not see cyrillic at all. for example a string has letters but compiler shows it is empty. when I use english letters it is ok. I also changed locale and it is now better in other programs but visual studio compiler does not understand...

Prof-Fansworth, 2017-03-11:
Hello.
I have the similar problem but I got some results solving it. As was said by VitaliGN when you use cyrilic letters in the code, they become broken on the screen after uploading program. I noticed then I insert my code in the example sketch which is located c:\Users\user\Documents\Arduino\libraries\U8g2\examples\full_buffer\HelloWorld\HelloWorld.ino cyrillic letters are NOT broken on the display, and in addition then I move my project from c:\Users\user\Documents\Arduino\project\project.ino to the path c:\Users\user\Documents\Arduino\libraries\U8g2\examples\full_buffer\project\project.ino cyrillic letters are OK on the screen. But I cannot save my project. IDE told me that the sketch is readonly and offer me to save it to another location. I think that compiler somehow modify the code from the sketch and then upload broken to the arduino. If you press "fix encoding and reload" in the IDE all cyrillic letters become broken in the code and they become look similar like broken letters on the display.

VitaliGN, 2017-03-12:
olikraus "-finput-charset=charset" should be done in cmd?

olikraus, 2017-03-12:
I think i was wrong and there is no easy way to pass this option to gcc.

VitaliGN, 2017-03-12:
Ok, I am the newbie for gcc and probably you gave me the first reason to think about what it is).
By the way Prof-Fansworth i checked pasting my code into HelloWorld, but it showed same result for me, there were no letters.
Some more links about this:
http://playground.arduino.cc/Code/UTF-8
http://playground.arduino.cc/Main/Utf8ascii

olikraus, 2017-03-12:
Maybe you should post some more complete examples. The font also has to fit and the .print command requires special setup.

VitaliGN, 2017-03-13:
olikraus, I am sorry, I think it is not connected directly to display library, it is about compiler, at the moment I opened an issue on VM forum, if you wish you could follow it here: http://www.visualmicro.com/forums/YaBB.pl?num=1489331854/1#1

olikraus, 2017-03-13:
ok, so we can close this issue?

VitaliGN, 2017-03-13:
Yes, I think so

closed 2017-03-26 by olikraus