olikraus/u8g2 #204: fntgrpnbp fonts installation

VitaliGN, 2017-03-01:
Hello olikraus,

I have found here some interesting fonts, (link below).
I am interested in these: 

u8g2_font_glasstown_nbp_tn

u8g2_font_mercutio_sc_nbp_t_all

u8g2_font_nine_by_five_nbp_t_all

they could solve my language fonts, please give me a link for a guide on how to install them

https://github.com/olikraus/u8g2/wiki/fntgrpnbp

olikraus, 2017-03-01:
Some related issues:
#170 
#54

olikraus, 2017-03-01:
Bdfconv is here:
https://github.com/olikraus/u8g2/blob/master/tools/font/bdfconv/Makefile

You need to compile this by yourself. I do not have access to a desktop PC at the moment.

VitaliGN, 2017-03-01:
I use Mega 2560 with OLED 1.3 Monochrome SH1106.
I am now trying to setup a library which is called simply u8g2. and when I type #include u8g2lib.h it cannot be found, and I have to retype the include as such: #include "u8g2-master\cppsrc\U8g2lib.h" in order to make it found. 
And here I have got one more question: here on the site there are two libraries of version 2 (u8g2 and u8g2-Arduino) Which would be correct to use? I suppose that Arduino version should be what I need, but there are no tools and additional fonts. Right now I am lost at the question on what should I move on further?

olikraus, 2017-03-01:
U8g2 can be installed via Arduino IDE library manager. Use U8g2lib.h for the include statement within Arduino IDE.

GitHub U8g2 is the development lib. U8G2-Arduino the version for the Arduino IDE.

VitaliGN, 2017-03-01:
A development lib means it is for a developer = you Oliver?

olikraus, 2017-03-01:
Well, yes, kind of..
U8G2-Arduino is derived from U8g2 lib.

VitaliGN, 2017-03-06:
I have a lot of reading and i did not make a good question yet, so I should close the issue temporarily until I compose it.

closed 2017-03-06 by VitaliGN