olikraus/u8g2 #205: Lack of SH1106 chip driver in compare to

VitaliGN, 2017-03-01:
As I understand the u8g2 library is a new version of u8glib. But I have an 1.3 inch OLED display which has a SH1106 chip driver with 4 pin I2C bus. So my issue is that u8glib has this constructor call initialization in line 103 (see link below), but u8g2 has only 2 SPI constructor calls in commented area of it's helloworld example. If I use SSD1306 then I have a corrupted line on the side of display.

https://github.com/olikraus/U8glib_Arduino/blob/master/examples/HelloWorld/HelloWorld.ino
line 103 = U8GLIB_SH1106_128X64 u8g(U8G_I2C_OPT_NONE);	// I2C / TWI

JamesGKent, 2017-03-01:
there are many constructors for SH1106 see here:
[https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#sh1106-128x64_noname](https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#sh1106-128x64_noname)

and below that are others depending on make and resolution.

VitaliGN, 2017-03-01:
Thank you I will try it when I get to my breadboard

olikraus, 2017-03-01:
Yes, the examples contain only a small fraction of the constructors.

VitaliGN, 2017-03-01:
I want to use this constructor U8G2_SH1106_128X64_NONAME_F_HW_I2C(rotation, [reset [, clock, data]]), but compiler highlights reset, clock and data by red color and saying that it is undefined, should I remove brackets and use SCL, SCK pins?

(thank you for your support, I will come back tomorrow)

olikraus, 2017-03-01:
Yes, replace this with Arduino pin numbers.

VitaliGN, 2017-03-02:
good morning,

U8G2_SH1106_128X64_NONAME_F_HW_I2C(rotation, reset, 21, 20);

seems no warnings, but I noticed that all constructors have "reset" parameter. but my I2C display has only VDD, GND, SCL, SDA. is it ok that there is a "reset" parameter?

olikraus, 2017-03-02:
If reset is not the, just use U8X8_PIN_NONE.

Note, that the i2c pins are usually fixed. This means, the last two arguments are only required for ESP based board s which allow any pins to be used.

VitaliGN, 2017-03-02:
I have just noticed that there are default values in the header files for each parameter, I think I start to understand. it is because it has a default value of "U8x8_pin_none". phew a little step further

VitaliGN, 2017-03-02:
I wrote right after you, ok, so considering what you wrote about last two arguments - I could leave only the rotation parameter? Oh, by the way the rotation parameter should be left as "rotation" or I should put some number as "0" there?

VitaliGN, 2017-03-02:
U8G2_SH1106_128X64_NONAME_F_HW_I2C(rotation, reset, 21, 20);
hm, now it again highlights commas saying that there should be ')' after "rotation" and numbers 21 and 20 saying that there should be identifier. a minute ago it was ok, now again.

JamesGKent, 2017-03-02:
the options for the rotation argument are here:
[https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#rotation](https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#rotation)

so assuming you want the default no rotation, and using hardware i2c your constructor should be:  
`U8G2_SH1106_128X64_NONAME_F_HW_I2C(U8G2_R0);

VitaliGN, 2017-03-02:
cool, I understand, if I use HW in constructor I logically do not need to put 21 and 20 pins because they should be defined by HW, and if I would want to change to another pins I should have used the SW instead of HW!

VitaliGN, 2017-03-02:
my mistake also was that I did not use the "u8g2" before brackets to initialize the display class

JamesGKent, 2017-03-02:
ah, i didn't spot that... so assuming all is working now are you happy to close this issue?

olikraus, 2017-03-02:
:-)

VitaliGN, 2017-03-02:
one more question - would it be possible to use two displays and make it so:
U8G2_SH1106_128X64_NONAME_F_HW_I2C Oled1(U8G2_R0);
U8G2_SH1106_128X64_NONAME_F_HW_I2C Oled2(U8G2_R0);
?
seems no mistake in compiler but I can not check it right now with displays

olikraus, 2017-03-02:
Yes, this is possible, but you need to assign different i2c client addresses. Usually there are solder bridges on the OLED to modify this address. In U8g2, there is a command to change the address.

VitaliGN, 2017-03-02:
Would it be ok to use it like that:
```
U8G2_SH1106_128X64_NONAME_F_HW_I2C Oled1(U8G2_R0);
U8G2_SH1106_128X64_NONAME_F_HW_I2C Oled2(U8G2_R0);

void setup()
{
	Oled1.begin();

	Oled2.setI2CAddress(0x78); // it has to be before the begin?
	Oled2.begin();
}
```

VitaliGN, 2017-03-02:
yes, Reference says that setting address should be before the begin() 
https://github.com/olikraus/u8g2/wiki/u8g2reference#seti2caddress

VitaliGN, 2017-03-02:
I am lacking questions about OLED initialization now, I will have to check all this in the evening with the board. hope now after such a great support I understand the main point and this issue can be closed

closed 2017-03-02 by VitaliGN

olikraus, 2017-03-02:
Let us know if you have further questions.